ARMmbed/mbed-os #5777: VK RZ A1H: needs an update from RZ A1H (cmsis, rtos related)

0xc0170, 2018-01-03:
### Description
- Type: Bug
- Priority: Major

### Bug

VK RZ A1H target requires an update. As it does not share the same codebase, but very identical implementation, it needs to be updated to be re-enabled. I had a quick look, fixed few errors there, and stopped at MMU and cache settings in the startup (cmsis updated their API, and RZ A1H changed few things in the startup). 

It would be much easier if they share as much common code as they could.

Related: https://github.com/ARMmbed/mbed-os/pull/5770 - this one fixes some issues we found.

cc @TomoYamanaka @mbedNoobNinja Please review and help

TomoYamanaka, 2018-01-04:
@0xc0170

Thank you for your cooporation.

@mbedNoobNinja

Could you acept to do this?
I addressed the updates regarding to RZ_A1H by #5628. By referencing with this, I believe that you can take a action of updates VK_RZ_A1H with comparative ease.

For your information, 
 I suggested the new folder structiures to share common codes previously(#4440).
 I will PR this changes together with adding our new target board(GR-LYCHEE) soon.

Therefore, After our this PR passed, I hope that you PR the changes regarding VK_RZ_A1H.
 Make a sense?

0xc0170, 2018-01-05:
> I will PR this changes together with adding our new target board(GR-LYCHEE) soon.

👍  If we can align those 3 at the same time, it would be good.

mbedNoobNinja, 2018-01-06:
Hi @TomoYamanaka
I actually was looking forward the PR from (#4440) and your board (GR-LYCHEE). I would be about to start comparing them after the PR in question is made. But if there is no such PR, what I should compare?
(#5628  & the current state of VK_RZ_A1H)?

TomoYamanaka, 2018-01-07:
HI @mbedNoobNinja 

Sorry to keep you waiting.
Can you go ahead your work by referring to this(#5628 & the current state of VK_RZ_A1H) until I PR?

TomoYamanaka, 2018-01-09:
Hi @mbedNoobNinja 

Thank you for waiting.
I did PR the registration of GR-LYCHEE target and the new folder structiures to share common codes(#5813). 
Could you PR the update of VK_RZ_A1H related after my PR merged?

mbedNoobNinja, 2018-01-09:
I will, after the (#5813) is merged, but my PR concerning the VK_RZ_A1H, may come a bit later. (with a week or two delay)

TomoYamanaka, 2018-01-10:
I look forward to your PR.

mbedNoobNinja, 2018-02-12:
Hi colleagues.
Do you have any idea why these paths:
```
\mbed-os\targets\TARGET_RENESAS
\mbed-os\targets\TARGET_RENESAS\TARGET_RZ_A1XX
\mbed-os\targets\TARGET_RENESAS\TARGET_RZ_A1XX\common
\mbed-os\targets\TARGET_RENESAS\TARGET_RZ_A1XX\TARGET_VK_RZ_A1H
\mbed-os\targets\TARGET_RENESAS\TARGET_RZ_A1XX\TARGET_VK_RZ_A1H\device
\mbed-os\targets\TARGET_RENESAS\TARGET_RZ_A1XX\TARGET_VK_RZ_A1H\device\inc
\mbed-os\targets\TARGET_RENESAS\TARGET_RZ_A1XX\TARGET_VK_RZ_A1H\device\inc\iobitmasks
\mbed-os\targets\TARGET_RENESAS\TARGET_RZ_A1XX\TARGET_VK_RZ_A1H\device\inc\iodefines
\mbed-os\targets\TARGET_RENESAS\TARGET_RZ_A1XX\TARGET_VK_RZ_A1H\device\TOOLCHAIN_ARM_STD
```
are not included in .includes_md5 when compiling anything for VK_RZ_A1H?

mbedNoobNinja, 2018-02-15:
I tried. The board was temporrary disabled by this
`"release_versions": []`, so I changed that to `"release_versions": ["2", "5"]`. But then the CLI complained
`[Error] irq_ctrl_gic.c@27,0:  #5: cannot open source input file "cmsis.h": No such file or directory`, because cmsis.h is in one of the missing paths above. I do not found any suspicuous cause in scripts in **tools/toolchain's**, which are responsible for constructing the .include_md5. I don't know where to look further.

0xc0170, 2018-02-15:
@mbedNoobNinja Can you provide your branch that we could reproduce the issue?

mbedNoobNinja, 2018-02-16:
Ok. Here it is: https://github.com/mbedNoobNinja/mbed/tree/CLI_issue

0xc0170, 2018-02-16:
@mbedNoobNinja that helped, the issue is with `extra_labels`, you are overwritting labels, not adding them . Use `extra_labels_add` that will fix the problem you are seeing

mbedNoobNinja, 2018-02-19:
Thank you @0xc0170, I will PR the changes along with the test results.

mbedNoobNinja, 2018-03-01:
Yeah, there it is **#6245** ... finally ! @TomoYamanaka, @0xc0170 thank you for the waiting. One RTSP client delayed me ... It turned out getting MJPEG stream from Ethernet & playing on MBED board, is not so easy, as I expected. By the way, Have you met a decent RTSP client for MBED (with the whole glory of RTCP and more specifically senders and receivers reports) or something close that should do the job ?

TomoYamanaka, 2018-05-22:
@0xc0170 Can I close this issue because this was updated by #6245 ?

closed 2018-05-24 by cmonr