ARMmbed/mbed-os #5778: REALTEK_RTL8195AM GPIO slow

LiyouZhou, 2018-01-03:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Enhancement
- Priority: Major

---------------------------------------------------------------
## Enhancement

**Target**
REALTEK_RTL8195AM

**Toolchain:**
GCC_ARM

**Toolchain version:**
7.2.1 20170904

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
`2b4ff78ab Merge pull request #5738 from ARMmbed/release-candidate`

**Description**
Toggling gpio on this platform is slow. It taks ~7us to toggle a pin. where as on platform such as k64f the pin toggle time is around 300ns which is 20x faster.

Slow GPIO makes bit banging impossible.  this driver https://os.mbed.com/users/bridadan/code/WS2812/ wouldn't work for example.

**Steps to reproduce**
This code prints out how much time it takes to toggle a pin:
```
#include "mbed.h"
#include "gpio_api.h"

#define GPIO_LED_PIN            PE_0

int main()
{
    gpio_t gpio_led;
    gpio_init(&gpio_led, GPIO_LED_PIN);
    gpio_dir(&gpio_led, PIN_OUTPUT);    // Direction: Output
    gpio_mode(&gpio_led, PullNone);     // No pull

    uint32_t num_of_loops = 10000;
    printf("start\r\n");

    for (uint8_t k = 0;  k < 255; k++)
    {
        us_ticker_init();
        __disable_irq();
        start = us_ticker_read();
        for (uint32_t i=0; i<num_of_loops; i++)
        {
            gpio_write(&gpio_led, 1);
            for(j=0; j<k; j++) __NOP();
            gpio_write(&gpio_led, 0);
            for(j=0; j<k; j++) __NOP();
        }
        end = us_ticker_read();
        __enable_irq();
        uint32_t ns_per_loop = (end-start)*1000/num_of_loops/2;
        printf("k = %u: %u ns per loop\r\n", k, ns_per_loop);

        if (ns_per_loop > 1000) break;
    }
```

LiyouZhou, 2018-01-03:
@ARMmbed/team-realtek

prashantrar, 2018-03-02:
@LiyouZhou 

The GPIO register operation clock is PCLK which is a 41.67MHz clock. So a GPIO toggling time should be under 1us. We can do a measurement with a scope to confirm this. However in this case, the measurement done by you is based on the time from us_ticker_read(). So the time seems is not precise since us_ticker_read() is counted by our G-Timer and our G-Timer resolution is 1/32K hence it will not be able to measure accurate switching times.

M-ichae-l, 2018-03-07:
Hi, @LiyouZhou 
The GPIO toggling time is around 600ns. For the fix for this issue, please refer to #6289 .

LiyouZhou, 2018-03-13:
Fantastic, I look forward to tring it out. Thank you for the fix.

M-ichae-l, 2018-05-22:
I believe this issue has already been solved by #6326. Is there any further issue or we can close this issue.

closed 2018-06-18 by 0xc0170