ARMmbed/mbed-os #5779: REALTEK_RTL8195AM us timer resolution too corse

LiyouZhou, 2018-01-03:
### Description
- Type: Enhancement
- Priority: Major

---------------------------------------------------------------
## Enhancement

**Target**
REALTEK_RTL8195AM

**Toolchain:**
GCC_ARM

**Toolchain version:**
7.2.1 20170904

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
`2b4ff78ab Merge pull request #5738 from ARMmbed/release-candidate`

**Description**
The us ticker on REALTEK_RTL8195AM uses a 32k clock source which means the resolution is ~30us. Hence the us ticker doesn't have us resolution. https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Realtek/TARGET_AMEBA/us_ticker.c#L30

@ARMmbed/team-realtek

Archcady, 2018-01-09:
Hello @LiyouZhou the g-timer clock source is 32K (32000 Hz), and there is no option for the clock source selection. Also, there is no way to adjust (over clock) the 32K freq. So the tick resolution is an hardware limitation.

samchuarm, 2018-01-12:
Hi @Archcady does this mean it would also affect the GPIO toggle time as described in [https://github.com/ARMmbed/mbed-os/issues/5778](url) ?

prashantrar, 2018-01-30:
Yes @samchuarm , It will affect GPIO toggle as well because this is a hardware limitation, we cannot give true microsecond precision using our g-timer.

0xc0170, 2018-01-30:
> Hello @LiyouZhou the g-timer clock source is 32K (32000 Hz), and there is no option for the clock source selection. Also, there is no way to adjust (over clock) the 32K freq. So the tick resolution is an hardware limitation.

There is no other timer that could be used by us ticker?

closed 2018-12-19 by ciarmcom