ARMmbed/mbed-os #5783: How to override mbed_sdk_init() for an EFM32 target?

amq, 2018-01-04:
### Description
- Type: Question

---------------------------------------------------------------

## Question

There is an EFM32-global `mbed_sdk_init()` definition in [TARGET_EFM32/common/mbed_overrides.c](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Silicon_Labs/TARGET_EFM32/common/mbed_overrides.c)

How can I override it for a specific target? For example, I'd like to skip `EMU_DCDCInit()`

0xc0170, 2018-01-05:
sdk_init is weakly linked but strongly defined for efm32 targets. It should provide the default target setup that an application can reconfigure if needed.

> How can I override it for a specific target? For example, I'd like to skip EMU_DCDCInit()

Can you provide details why this would be skipped? Would it make sense to be configurable (not applicable for all targets , or not? )

@ARMmbed/team-silabs

amq, 2018-01-06:
The problem is that custom targets may require different settings for `EMU_DCDCInit()`. In some cases (such as mine), the board won't reach `main()` at all, so I can't override the settings in application.

0xc0170, 2018-01-08:
@ARMmbed/team-silabs Can the startup seq be customized ? Please review t he proposal

stevew817, 2018-01-08:
@amq do you have a custom target based off of EFM32?

EMU_DCDCInit() should be called in all cases, but you are correct that the settings for the DCDC converter are dependent on the target board. The default values are correct for our boards, but might not be for custom boards.
Would you be okay with defining the DCDC settings in the target-specific device_peripherals.h header, in the same way we're defining the HFXO settings?

amq, 2018-01-08:
> Would you be okay with defining the DCDC settings in the target-specific device_peripherals.h header, in the same way we're defining the HFXO settings?

Absolutely.

stevew817, 2018-01-15:
@amq Please see PR #5855

stevew817, 2018-01-17:
@cmonr or @0xc0170 Please close this ticket as the requested feature was implemented by #5855

closed 2018-01-17 by 0xc0170

cmonr, 2018-01-17:
@stevew817 For reference, using a specific keywork in the PR will automatically close the issue as well: https://github.com/blog/1506-closing-issues-via-pull-requests