ARMmbed/mbed-os #5785: BLE builds cannot find mbedtls feature

hasnainvirk, 2018-01-04:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Related issue: `#5711`
- Priority: Blocker

---------------------------------------------------------------
## Bug

It seems the build system is failing to look for even omni-present features. 

**Target**
NRF51_DK|??

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-os:**
master

**Steps to reproduce**
Try building any BLE example alongwith including mbedtls/aes.h

0xc0170, 2018-01-04:
@pan- Have you experienced this? There's lorawan PR where had any example ble failure , as one of the headers in the paths included `aes.h` header file

0xc0170, 2018-01-05:
I found the cause. The BLE examples ignore mbedtls ! This is causing a problem if there is a code in mbed OS that needs mbedtls (tls is not an optional component?).

This is known limitation, reported here https://github.com/ARMmbed/mbed-os-example-ble/issues/45

It has been for over 2 years. 

@paul-szczepanek-arm  would you know a reason for this as @pan- is away this week

pan-, 2018-01-08:
@hasnainvirk This has been fixed in ble examples.

closed 2018-01-09 by 0xc0170