ARMmbed/mbed-os #5786: STM ticker not compatible with new ticker_api

zh0umu, 2018-01-04:
### Description
- Type: Bug 
- Priority: Blocker
new ticker_api add get_info in ticker_interface_t which use 32bit tick timer as defaut,however stm32 use 16bit timer.This casue Timer.read_ms() return wrong result.
---------------------------------------------------------------
## Bug

**Target**
STM32

0xc0170, 2018-01-04:
@ARMmbed/team-st-mcd please review

jeromecoutant, 2018-01-04:
Hi
This issue needs more details...

- which target are you using (most of STM32 are using 32 bits timer) ?
- could you add the code you are using in order to reproduce ?

Thx

LMESTM, 2018-01-04:
@zh0umu  @0xc0170  When moving from one ticker implementation to another, aren't you going to port the targets that are in place already ?

zh0umu, 2018-01-04:
I see this in mbed_retarget.cpp
// temporary - Default to 1MHz at 32 bits if target does not have us_ticker_get_info
MBED_WEAK const ticker_info_t* us_ticker_get_info()
{
    static const ticker_info_t info = {
        1000000,
        32
    };
    return &info;
}

jeromecoutant, 2018-01-04:
Yes, this looks correct, as even for 16bit STM32 targets, timer implementation is 32 bits.

zh0umu, 2018-01-04:
You are right. @jeromecoutant

closed 2018-01-04 by zh0umu