ARMmbed/mbed-os #5790: STM32F4: lp ticker failures on master

0xc0170, 2018-01-05:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

Reference: https://github.com/ARMmbed/mbed-os/pull/5768#issuecomment-355425482

There are at least 2 failures from yesterday, the report provides binaries and test failures

@ARMmbed/team-st-mcd please review

jeromecoutant, 2018-01-05:
Hi
Do you have several test benches with NUCLEO_F401 ?
To be sure, can you unplug all these boards and make the test again ?

0xc0170, 2018-01-05:
> To be sure, can you unplug all these boards and make the test again ?

We can. How that would help (what is causing these failures) ?

jeromecoutant, 2018-01-05:
RTC configuration has been updated in a recent PR.
But as long as the board is powered, the new configuration is not applied as we checked before RTC init procedure if RTC has already been configured.

0xc0170, 2018-01-05:
> Do you have several test benches with NUCLEO_F401 ?
To be sure, can you unplug all these boards and make the test again ?

@studavekar @kegilbert all of the boards were reset previously?

0xc0170, 2018-01-05:
Affected PRs:

- https://github.com/ARMmbed/mbed-os/pull/5708

0xc0170, 2018-01-05:
> RTC configuration has been updated in a recent PR.

What is the PR, can you reference it ? The git log for the RTC API:  https://github.com/ARMmbed/mbed-os/commits/master/targets/TARGET_STM/rtc_api.c .  I recall we had similar error a while ago, and devices were reseted . Is there a  way to make this backward compatible ? (no hard reset target needed)

To me, this seems like the problem is back, thus might be a bug somewhere?

jeromecoutant, 2018-01-05:
Is it possible that CI is sometimes testing MBED branch started from an "old" commit ?

0xc0170, 2018-01-05:
> Is it possible that CI is sometimes testing MBED branch started from an "old" commit ?

That can be (although rebase is always good 😈 ). The affected PR were based on the recent master. 

In case there was PR on the old commit, should not lead to a breakage like this? One test just finished few minutes ago, the tests were green

0xc0170, 2018-01-08:
Just got two F4 lp ticker failures here https://github.com/ARMmbed/mbed-os/pull/5597#issuecomment-355995753

studavekar, 2018-01-12:
Ticker failure on F7 not sure if its related  https://github.com/ARMmbed/mbed-os/pull/5795#issuecomment-357342685 & https://github.com/ARMmbed/mbed-os/pull/5760#issuecomment-357329201

0xc0170, 2018-01-15:
#5841 also

@jeromecoutant I'll reproduce today this ticker tests locally with the latests tests that run. 
We might need to disable lp ticker for those devices if not fixed soon.

mprse, 2018-01-15:
I tried to reproduce failure of the lp ticker tests on NUCLEO-F429ZI board without success.

 Tried to run the following tests:
- tests-mbed_hal-lp_ticker,
- tests-mbed_drivers-lp_timer

several times using GCC_ARM, ARM and IAR tool-chains and develop profile with the following additional flags (tests on CI are compiled with these flags):
- "-DMBED_TRAP_ERRORS_ENABLED=1",
- "-DMBED_HEAP_STATS_ENABLED=1",
- "-DMBED_STACK_STATS_ENABLED=1".

No failure has been detected.

Performed also the following test:
- created temp branch with head commit from 5 September, 
- pulled out usb cable and connect it back, 
- run lp ticker test (passed), 
- switch back to master and run the test again (passed x3).

jeromecoutant, 2018-01-16:
@0xc0170 
I think I could find some optimization in the code execution.
Tests on going

jeromecoutant, 2018-01-24:
What is the status after 1 week test ?
Thx

0xc0170, 2018-01-24:
Looks like it is fixed, I'll close it

closed 2018-01-24 by 0xc0170