ARMmbed/mbed-os #5793: Abstraction/virtualization of basic peripheral classes

steve-martin-mbed, 2018-01-05:
### Description
- Type: Enhancement
- Priority: Minor

## Enhancement

As suggested in the forum [,](https://os.mbed.com/forum/bugs-suggestions/topic/28469/) I'm opening up this for discussion here.

> As the availability of peripheral and driver variants continues to explode on the mbed platform, it becomes more and more difficult to write higher-level functionality that generically takes advantage of what is available. Polymorphism has been implemented in various parts of the code (i.e. network stack) to address this very issue. However, basic peripherals remain lacking. Moving to abstract class definitions (i.e. I2CBase, SPIBase, DigitalOutBase, etc...), or at least virtualization of basic functionality can ease this problem.  ...

0xc0170, 2018-01-05:
Can you give an example , lets take I2C class. What would I2C base provide, and then I2c also.

steve-martin-mbed, 2018-01-08:
I imagine the base class would define most of the functions already contained in the existing I2C class, but provide for a disconnection (through virtualization) from the MCU hardware.  For instance, right now, I2C::read, I2C::write, I2C::start, etc... call directly the MCU driver libraries i2c_read, i2c_write, i2c_start, etc.. respectively.  If I want to use a device (ie. an I2C display) using a MCU I2C port, all is great.  But if I need to drive the display with a SoftI2C, now I must change the display driver which is hard-coded to the I2C class.  If the I2C class was virtualized or derived from an abstract base class, then the SoftI2C could override the functionality of read, write, stop, etc..., and be referenced as the I2C base class, removing the need to modify the display driver.

An even better example would be a class for an I2C-based port expander.  Virtualizing the I2C interface would not only allow the port expander driver be driven by different types of I2C interfaces, but the port expander itself could also provide virtualized IO ports derived from abstracted DigitalIn, DigitalOut, etc.., to expand the functionality of the platform without requiring rewrites of higher-level drivers.

Whether defining an abstract [I2C] base class or simply virtualizing the functions of the [I2C] class [and calling it the base class] is to be determined.  In my limited experience, I would think that an abstract base class defining the minimal operation of any derived I2C class would be the cleanest.  But, maybe there are caveats that other team members are aware of that would suggest not using an abstract base class.  The current set of I2C methods seems to be mostly adequate as a basic functional set.  I would add a constructor that allows defining the SDA and SCL pins by Digital reference in addition to the one that defines by PinName.  This would allow usage of I/O pins of port expanders to function as SDA and SCL.  The only missing base function that I might argue for would be a transfer function that writes, restarts and then reads, but this is easy enough to implement using the write and read methods.

steve-martin-mbed, 2018-01-08:
I happen to be playing around with the I2C class right now.  Just to see if it works (and see how messy it is), I'll try making an abstract base class, then derive I2C and SoftI2C from that.

0xc0170, 2018-01-08:
cc @bulislaw @SenRamakri @c1728p9 @scartmell-arm

SenRamakri, 2018-12-04:
Hello @steve-martin-mbed - Thanks for the suggestion and we do think its a great idea. Unfortunately we don't have any immediate plans to to abstract the peripheral classes. But feel free to provide an implementation and we would be happy to review and integrate. At present, we are closing this issue.

closed 2018-12-04 by SenRamakri