ARMmbed/mbed-os #5796: ARM: ARMC6: Linker errors due to cpu not being specified in the scatter file

hosse005, 2018-01-05:
### Description
- Type: Bug
- Related issue: Not known
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
ALL

**Toolchain:**
ARM

**Toolchain version:**
5.06 and 6.8

**Expected behavior**
Targets with a preprocessor defined in the scatter file should link successfully

**Actual behavior**
Targets with a preprocessor defined in the scatter file fail to link with the ARM compiler

**Steps to reproduce**
1.  Pull the mbed-os-example-blinky app
2.  Import the latest mbed-os (>= mbed-os-5.5.4)
3.  Compile with either the ARM or ARMC6 compiler and target the MultiTech xDot
(e.g. mbed compile -t ARMC6 -m xdot_l151cc)

0xc0170, 2018-01-08:
Is this related to https://github.com/ARMmbed/mbed-os/issues/5421 ?

hosse005, 2018-01-08:
Yes I believe so.

PR #5797 (2018-02-07): ARMC5+6: Specify CPU for ARM scatter file preprocessor
Fixes #5796

## Description

Fix for targets which include the c preprocessor in the ARM and ARMC6 scatter file.

## Status

**READY**

## Migrations

NO

## Steps to test or reproduce
1. Pull the mbed-os-example-blinky app
2. Import the latest mbed-os (>= mbed-os-5.5.4)
3. Compile with either the ARM or ARMC6 compiler and target the MultiTech xDot
(e.g. mbed compile -t ARMC6 -m xdot_l151cc)
4. Verify the app builds
commit 453a1d79 (2018-02-07): ARM: ARMC6: Specify CPU for ARM scatter file preprocessor

Fixes #5796
closed 2018-02-07 by cmonr