ARMmbed/mbed-os #5798: having problem when i2c transmision is interrupted by Ticker or os thread

zh0umu, 2018-01-06:
### Description
- Type: Question
- Priority: Minor
I'm using i2c to talk to MPU9250 at rate of 100hz with large data.When I use Ticker only to change a volatile variable, the bus lockup after some time randomly,with SCL high and SDA low.After lock up,reset only the mcu still not solve the bus locking problem.If  remove the slave ,bus recover.I believe it's the slave driving the sda low.
The same problem came across when using OS thread.
I'm looking to how the i2c driver implemented when interruptted.Glad to see if anyone can help

**Target**
STM32F103RB

0xc0170, 2018-01-08:
I2C is thread safe, thus the transaction can be interrupted. Can you share a code snippet to illustrate the issue? What is handled in the main thread, other threads and interrupts?

cmonr, 2018-01-23:
@zh0umu ^^^

zh0umu, 2018-01-25:
I suspend some register read instruction in HAL layer performing  unnoraml if interrupted ,I have no time right now .So I close it now and report later.

closed 2018-01-25 by zh0umu