ARMmbed/mbed-os #5799: Assert when reading character from serial port in callback - "not allowed in ISR context"

saedelman, 2018-01-06:
### Description

I am getting a "Mutex 0x200077d4 error -6: Not allowed in ISR context" assert when calling pc.getc() in the serial callback. I am using mBed version 158 (5.7.2)

pc.attach(&data,mbed::SerialBase::RxIrq);

void data(void)
{
char ch = pc.getc(); <-- GENERATES ASSERT
}

- Type: Bug 
- Priority: Blocker 

---------------------------------------------------------------
## Bug

**Target**
NRF52

**Toolchain:**
GCC_ARM

abougouffa, 2018-01-07:
Hi @saedelman, this not seems to be a bug, the `getc()` method like other standard `stdio` functions use `Mutex`es to protect the access to the standard output resource, and like is mentioned in the [API documentation](https://os.mbed.com/docs/v5.7/reference/mutex.html)
> The Mutex methods cannot be called from interrupt service routines (ISR). In the current version of Mbed OS, if you attempt to use a mutex from within an ISR, nothing happens; attempts to lock a mutex succeed immediately, regardless of whether the lock is actually free. In other words, if you acquire a mutex lock in an ISR, you can break the thread safety mechanisms and introduce race-conditions into an otherwise safe piece of code. Future versions of Mbed OS will provide warnings and ultimately prevent this from happening. 

As a workaround, you can set a flag (a variable) in the interrupt routine, and in the main loop; check this flag and do what you want safely.

`bool interrupt_invoked = false;

void data(void)
{
    interrupt_invoked = true;
}

int main()
{
    ....
    ....
    while (1) { // main loop
        ....
        ....
        if (interrupt_invoked) {
            interrupt_invoked = false;
            char ch = pc.getc();
        }
    }
}
`

saedelman, 2018-01-07:
Yeah, my bad. I'm upgrading from mBed OS 5.4.7 and the documentation for mBed is lagging behind as the examples online still show doing the getc() in the ISR. I've switched to using RawSerial for now, so we'll close this issue.

closed 2018-01-07 by saedelman