ARMmbed/mbed-os #5801: Allow mbed_app.json to override OS_IDLE_THREAD_STACK_SIZE for Nordic targets

saedelman, 2018-01-08:
### Description
- Type: Bug 
- Priority:  Minor

---------------------------------------------------------------
## Bug

The iOS_IDLE_THREAD_STACK_SIZE define  is hard coded for Nordic targets in:

mbed-os/targets/TARGET_NORDIC/mbed_rtx.h 

This should be wrapped in #ifndef/#endif otherwise the user defined setting in mbed_app.json is ignored.

**Target**
NRF52_DK

**Toolchain:**
GCC_ARM|

**Toolchain version:**

**mbed-cli version:** 
1.1.1

## Enhancement

It would be nice if the idle thread and timer threads were named so that they are easily identified when dumping threads stats (using mbed-memory-status for example).

0xc0170, 2018-01-08:
I checked, there are at least 2 platforms that define this macro, none use ifndef.

> It would be nice if the idle thread and timer threads were named so that they are easily identified when dumping threads stats (using mbed-memory-status for example).

For this, I assume instead of NULL, there could be a name:

```
static const osThreadAttr_t os_idle_thread_attr = {
  NULL,
  osThreadDetached,
  &os_idle_thread_cb,
  (uint32_t)sizeof(os_idle_thread_cb),
  &os_idle_thread_stack,
  (uint32_t)sizeof(os_idle_thread_stack),
  osPriorityIdle,
  0U, 0U
};
```

@pan- @ARMmbed/team-nordic

0xc0170, 2018-01-08:
For the names in idle thread and timer threads, can you create an issue in CMSIS 5: https://github.com/ARM-software/CMSIS_5/ with details , to understand the use case.  It could be configurable (default NULL as it is).

pan-, 2018-01-08:
@bulislaw Is there any plan to allows compile time configuration of the RTOS ?

bulislaw, 2018-01-09:
I would say it makes sense to have the sizes configurable for timer and idle thread stacks. And mbed_app.json would be the right place. @saedelman would you be able to create PR for that?

JonatanAntoni, 2018-01-23:
We added some defines to add names to idle and timer threads, see [aa7a332](https://github.com/ARM-software/CMSIS_5/commit/aa7a332938feb50f2e9d3d998c8ced442445ffd2).

saedelman, 2018-01-23:
@JonatanAntoni Thank you. 

@bulislaw Agreed. If the RTOS allows me to define my own idle function (e.g. Thread::attach_idle_hook()) then I should be able to define the stack for it by overriding OS_IDLE_THREAD_STACK_SIZE, or it will most certainly lead to an overflow.  I'm happy to create a PR, can you clarify how I do that? Not familiar.

bulislaw, 2018-01-24:
@saedelman Hava a look how main thread stack is configured here: https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/mbed_rtx_conf.h#L31 it'll be the same for the other stacks.

marcuschangarm, 2018-05-09:
@saedelman 

The hard coded `OS_IDLE_THREAD_STACK_SIZE` has been removed from `mbed-os/targets/TARGET_NORDIC/mbed_rtx.h` and shouldn't interfere with your application anymore: https://github.com/ARMmbed/mbed-os/pull/6845