ARMmbed/mbed-os #5803: New test's directory is not under .mbedignore

TaniaMirzin, 2018-01-08:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Bug 
- Minor

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**mbed-os sha:**
eca67ca7dafab4ef70c21e2463b541132d0dd691
release 5.7.2

**Expected behavior**
All tests directories is under .mbedignore
**Actual behavior**
The new directory mbed-os\features\filesystem\littlefs\TESTS_COMMON is not uder .mbedignore

Please consider to add this path to https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/littlefs/.mbedignore

https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/littlefs/.mbedignore

0xc0170, 2018-01-08:
@geky As for now, there would be some copy-paste sharing for those tests, there is an opened issue how to handle this better as I recall. Can you please make sure the test code is not part of the app build

TaniaMirzin, 2018-01-08:
@shelib01

geky, 2018-01-09:
related issue https://github.com/ARMmbed/mbed-os/issues/5524
proposed fix https://github.com/ARMmbed/mbed-os/pull/5819

Unfortunately it isn't as simple as adding the TEST_COMMON to .mbedignore, since that would prevent the TEST_COMMON from being used in tests. The test framework we are using currently doesn't have a home for common code.

We could copy-paste these tests as @0xc0170 mentions. IMO copy-paste is not a good solution, especially since these tests may develop quickly with all the recently introduced storage code.

closed 2018-01-18 by cmonr