ARMmbed/mbed-os #5804: CLI export to SW4STM32 fails on OSX

Hoel, 2018-01-08:
### Description
- Type: Bug 
- Priority: Blocker 

---------------------------------------------------------------
## Bug

**Target**
ALL

**Toolchain:**
GCC_ARM

When i build blinky project with CLI everything works fine, when i try to export the same project for SW4STM32 it fails. This is a custom target (LORAM3_L151C8) but i also tried with already include targets, same result. The mbed blinky project has been downloaded on 19/12 from CLI.

iMac-de-Hoel-2:mbed-os-example-blinky hoel$ mbed compile -m LORAM3_L151C8 -t GCC_ARM
Building project mbed-os-example-blinky (LORAM3_L151C8, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Elf2Bin: mbed-os-example-blinky
+------------------+-------+-------+------+
| Module           | .text | .data | .bss |
+------------------+-------+-------+------+
| [fill]           |    56 |     8 |    8 |
| [lib]/c.a        | 24364 |  2204 |   56 |
| [lib]/gcc.a      |  4512 |     0 |    0 |
| [lib]/m.a        |    80 |     0 |    0 |
| [lib]/misc       |   228 |    16 |   28 |
| main.o           |    72 |     4 |   28 |
| mbed-os/drivers  |   136 |     4 |  100 |
| mbed-os/hal      |   889 |     0 |   26 |
| mbed-os/platform |  1384 |     4 |  270 |
| mbed-os/rtos     |  8457 |   180 | 5988 |
| mbed-os/targets  |  7082 |     4 |  416 |
| Subtotals        | 47260 |  2424 | 6920 |
+------------------+-------+-------+------+
Total Static RAM memory (data + bss): 9344 bytes
Total Flash memory (text + data): 49684 bytes

Image: ./BUILD/LORAM3_L151C8/GCC_ARM/mbed-os-example-blinky.bin
iMac-de-Hoel-2:mbed-os-example-blinky hoel$ mbed export -i sw4stm32 -m LORAM3_L151C8
Scan: .

Create a System Workbench for STM32 managed project
Project name: mbed-os-example-blinky
Target:       LORAM3_L151C8
Toolchain:    GCC_ARM

Symbols: 40
Include folders: 58
Source folders: 30, with 58 exclusions
Exclude folders: 58
Linker script:   mbed-os/targets/TARGET_STM/TARGET_STM32L1/TARGET_LORAM3_L151C8/device/TOOLCHAIN_GCC_ARM/STM32L151XC.ld
Traceback (most recent call last):
  File "/Users/hoel/mbed-os-example-blinky/mbed-os/tools/project.py", line 259, in <module>
    main()
  File "/Users/hoel/mbed-os-example-blinky/mbed-os/tools/project.py", line 254, in main
    build_profile=profile, app_config=options.app_config)
  File "/Users/hoel/mbed-os-example-blinky/mbed-os/tools/project.py", line 96, in export
    app_config=app_config)
  File "/Users/hoel/mbed-os-example-blinky/mbed-os/tools/export/__init__.py", line 342, in export_project
    macros=macros)
  File "/Users/hoel/mbed-os-example-blinky/mbed-os/tools/export/__init__.py", line 207, in generate_project_files
    exporter.generate()
  File "/Users/hoel/mbed-os-example-blinky/mbed-os/tools/export/sw4stm32/__init__.py", line 519, in generate
    'board_name': self.BOARDS[self.target.upper()]['name'],
KeyError: 'LORAM3_L151C8'
[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /Users/hoel/mbed-os-example-blinky/mbed-os/tools/project.py -i sw4stm32 -m LORAM3_L151C8 --source ." in "/Users/hoel/mbed-os-example-blinky"
---
iMac-de-Hoel-2:mbed-os-example-blinky hoel$

0xc0170, 2018-01-08:
Based on the errors in the traceback, seems that board is not in the list, the exception might be handled to provide a log that a board is not supported (can help during offline development).

To add a support, you should add your target to BOARDS dictionary.

@ARMmbed/team-st-mcd

Hoel, 2018-01-08:
> To add a support, you should add your target to BOARDS dictionary.
It is in targets.json and has a folder.
also as told **I tried other existing targets (XDOT_L151CC and others)** with the same result, anyway the custom target is in the list, has a folder, can compile and the binary works.

```
iMac-de-Hoel-2:mbed-os-example-blinky hoel$ mbed export -i sw4stm32 -m XDOT_L151CC
Scan: .

Create a System Workbench for STM32 managed project
Project name: mbed-os-example-blinky
Target:       XDOT_L151CC
Toolchain:    GCC_ARM

Symbols: 40
Include folders: 58
Source folders: 30, with 58 exclusions
Exclude folders: 58
Linker script:   mbed-os/targets/TARGET_STM/TARGET_STM32L1/TARGET_XDOT_L151CC/device/TOOLCHAIN_GCC_ARM/STM32L151XC.ld
Traceback (most recent call last):
  File "/Users/hoel/mbed-os-example-blinky/mbed-os/tools/project.py", line 259, in <module>
    main()
  File "/Users/hoel/mbed-os-example-blinky/mbed-os/tools/project.py", line 254, in main
    build_profile=profile, app_config=options.app_config)
  File "/Users/hoel/mbed-os-example-blinky/mbed-os/tools/project.py", line 96, in export
    app_config=app_config)
  File "/Users/hoel/mbed-os-example-blinky/mbed-os/tools/export/__init__.py", line 342, in export_project
    macros=macros)
  File "/Users/hoel/mbed-os-example-blinky/mbed-os/tools/export/__init__.py", line 207, in generate_project_files
    exporter.generate()
  File "/Users/hoel/mbed-os-example-blinky/mbed-os/tools/export/sw4stm32/__init__.py", line 519, in generate
    'board_name': self.BOARDS[self.target.upper()]['name'],
KeyError: 'XDOT_L151CC'
```

```
[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /Users/hoel/mbed-os-example-blinky/mbed-os/tools/project.py -i sw4stm32 -m XDOT_L151CC --source ." in "/Users/hoel/mbed-os-example-blinky"
```

0xc0170, 2018-01-08:
https://github.com/ARMmbed/mbed-os/blob/master/tools/export/sw4stm32/__init__.py#L36 - this is where those targets should be added to enable the exporter

Hoel, 2018-01-08:
OK thank you it works now indeed. 
I cannot find the project folder, where is it generated by default?

0xc0170, 2018-01-08:
It should work out of the box, are you using your own GCC ARM installation or provided by stm32 ide?

@ARMmbed/team-st-mcd Do you recognize the errors above?

adustm, 2018-01-08:
Dear @Hoel 

I cannot help for your private target, but I've tried the closest STM device, the STM32F303VC that is on the DISCO_F303VC.
I did:
`mbed export -i sw4stm32 -m DISCO_F303VC`
Then I've imported the directory mbed-os-example-blinky in System Workbench _(File... Import... from an existing project... select the directory)_ and builded the project. It compiles and link prefectly.
```
Generating binary and Printing size information:
arm-none-eabi-objcopy -O binary "mbed-os-example-blinky-debug.elf" "mbed-os-example-blinky.bin"
arm-none-eabi-size -B "mbed-os-example-blinky-debug.elf"
   text	   data	    bss	    dec	    hex	filename
  26032	   2316	   7228	  35576	   8af8	mbed-os-example-blinky-debug.elf
```
Could you also try that and tell us if it works for you ?

Hoel, 2018-01-08:
@0xc0170 
i use the GCC ARM provided by SW4STM32 indeed. Also the projects i export from mbed online  (as SW4STM32 project) are imported and working fine in SW4STM32 (after small project properties tweaking).
The SW4STM32 project generated by CLI also has broken project properties, when i open them i got errors in SW4STM32 and i cannot access the C/C++ build settings at all which is weird since rojects exported from online compiler are working fine.

[image]
[image]
[image]
@adustm 
I just tried with DISCO_F303VC and it worked, actually i used this very folder to make mine (attached), what could be wrong then? 
[TARGET_LORAM3_F303CC.zip](https://github.com/ARMmbed/mbed-os/files/1612582/TARGET_LORAM3_F303CC.zip)

Hoel, 2018-01-09:
thank you, ok i try and report here

adustm, 2018-01-09:
I'm afraid you will need support from System Workbench. You could ask your questions on their forum here www.openstm32.org ?
Maybe the generated makefiles are not well generated. You could compare them with the one generated for the DISCO_F303VC... 

kind regards

Hoel, 2018-01-09:
I will ask them, not sure they will deal the problem, it looks like an mbed CLI issue. 
The projects exported from online mbed compiler are working fine, they dont have these run/debug errors.
The project exported from CLI with existing target :  mbed export -i sw4stm32 -m DISCO_F303VC
are also ok for run/debug, of course i cannot use it for actual target since MCU is not the same variant.

> Maybe the generated makefiles are not well generated.

Makefiles are generated by the mbed CLI not SW4STM32 right?
Only project exported from CLI with the custom target does exhibit the issue, how can this be related to SW4STM32? I think the problem is how the CLI handles custom targets, though custom target folder has been made by copying an existing target (DISCO_F303CV in this case).

Hoel, 2018-01-15:
follow up on the issue, i confirm the prblem is on CLI export side, i have been able to setup a working project by exporting with 
```
        'LORAM3_F303CC':
        {
            'name': 'STM32F3DISCOVERY',
            'mcuId': 'STM32F303CCT6'
        },
```
Which use STM32F3DISCOVERY' as target instead of my custom target (made from STM32F3DISCOVERY directory actually), when i use this the project is working fine in SW4STM32, but i have a to make numerous change in order to set the correct target afterwards in SW4STM32, i have to set the target manually in project properties. So if anyone can have a check on what is wrong in the exporter, to reproduce the issue, just make a custom target by copying the STM32F3DISCOVERY folder, add custom target in the __init__.py  dictionnary as mentioned before and in targets.json and then export SW4STM32 project and try to open it in SW4STM32, it wil fail because the project properties are broken. When you change the target definition with 
```
        'LORAM3_F303CC':
        {
            'name': 'STM32F3DISCOVERY',
            'mcuId': 'STM32F303CCT6'
        },
```
then the SW4STM32 project is working fine, this is definitely an issue with the CLI exporter, maybe there is another hardcoded dictionarry somwhere else where i should add my custom target?

bcostm, 2018-01-16:
For me the SW4STM32 is working fine. The problem for you is that SW4STM32 does not support custom boards which is normal I think...

Hoel, 2018-01-16:
does SW4STM32 tool stand for SW4STM32 IDE or for mbed CLI SW4STM32 exporter? SW4STM32 does support custom target, we can create it from scratch with no issue in the target menu, and actually i have the LORAM3_F303CC in my targets tabs since i created it.

jeromecoutant, 2018-01-24:
Hi

> after more tries, i got it to work

I think we can close the issue ?
Thx

closed 2018-01-24 by 0xc0170

Hoel, 2018-01-24:
Hello,
Sorry i havent worked on this lately, yes the export settings issue can be closed, problem is fixed by making sure custom target is created in SW4STM32 prior importing project and by adjusting project properties afterward (include path, config.h path, and few preprocessor defines).

There is still a big difference between a project exported from online compiler and from CLI, online compiler exported project result is including the mbed folder (obtained after building mbed-os with the specific target) which is small size and compile time is normal, CLI exported project includes the (huge +400mo) whole mbed-os folder and compile time is very long.
Thanks