ARMmbed/mbed-os #5805: Serial example with callback doesn't work on REALTEK_RTL8195AM

jarlamsa, 2018-01-08:
### Description
- Type: Bug
- Priority: Major
---------------------------------------------------------------
## Bug

**Target**
REALTEK_RTL8195AM

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
`eca67ca7d`

**Expected behavior**
User is able to write to and read from serial

**Actual behavior**
User is not able to write or read from serial

**Steps to reproduce**
Copy example three from
(https://os.mbed.com/docs/v5.7/reference/serial.html)

Remove led stuff. Compile, flash and run.

0xc0170, 2018-01-08:
@ARMmbed/team-realtek Please review

saedelman, 2018-01-08:
Two things to check, you need to use "RawSerial" not "Serial" constructors when using the getc() in the attach callback and you likely can't use printf() either. So, try compiling your code again with "Serial" replaced with "RawSerial" and remove the printf() but leave the led stuff so you can see that it is receiving data.

jarlamsa, 2018-01-09:
@SeppoTakalo

jarlamsa, 2018-01-09:
Replaced `Serial` with `RawSerial` and the old printf with
`pc.putc(pc.getc());`
Still not getting anything on the terminal screen.

0xc0170, 2018-01-09:
@jarlamsa Any other platform working ? 

I filed a bug for Serial three example as it is incorrect (filled in Handbook)

jarlamsa, 2018-01-09:
@0xc0170 The Serial example is working at least with K64F. Also RawSerial is working with K64F.

REALTEK_RTL8195AM is able to print out TX to the terminal, but fails to echo the RX back to the terminal.

jarlamsa, 2018-01-10:
Tested the Serial and RawSerial with DISCO_L475VG_IOT01A and UBLOX_EVK_ODIN_W2, working without problems.

Also the REALTEK_RTL8195AM works without problems if I remove the callback and just poll for characters in the while loop in main function. So the callback breaks the functionality on REALTEK_RTL8195AM.

jarlamsa, 2018-01-15:
@saedelman Are you investigating this?

saedelman, 2018-01-15:
No, I am not. The example 3 that you are referring to does not work on any target when using mbed-os above 5.4.7 because mutexes are not permitted inside an ISR and Serial is stream based and therefore wrapped in mutexes under the hood. You have to use "RawSerial pc(USBTX,USBRX)" instead of "Serial pc(USBTX,USBRX)" in the declaration to make it work. This is an ongoing problem with the documentation being out-of-sync.

0xc0170, 2018-01-15:
> No, I am not. The example 3 that you are referring to does not work on any target when using mbed-os above 5.4.7 because mutexes are not permitted inside an ISR and Serial is stream based and [...]

We fixed the example, need to push the new SHA to the docs

jarlamsa, 2018-01-15:
That's weird since I tested with other targets and they were working with that example.

saedelman, 2018-01-15:
It may depend on the hardware specific drivers as well, but you should not be able to use printf() in an ISR (as the example showed) with any version of mbed-os above 5.4.7 as it will immediately fault with osErrorISR: "Mutex xxxx error -6: not allowed in ISR context". Have you tried using RawSerial instead? The interface is identical to Serial and should work.

jarlamsa, 2018-01-16:
Yes, this is what I used with RawSerial.

```
#include "mbed.h"

DigitalOut led1(LED1);
DigitalOut led2(LED2);

RawSerial pc(USBTX, USBRX);

void callback_ex() {
    // Note: you need to actually read from the serial to clear the RX interrupt
    pc.puts("\r\ngot:");
    pc.putc(pc.getc());
    led2 = !led2;
}

int main() {
    pc.attach(&callback_ex);
    
    while (1) {
        led1 = !led1;
        wait(0.5);
    }
}
```

This is working with other targets, but not with the Realtek.

jarlamsa, 2018-01-17:
@saedelman could you check the provided RawSerial example?

saedelman, 2018-01-17:
I do not have access to a REALTEK_RTL8195AM target so cannot test directly, but your example should work. Have you tried compiling with the debug profile (e.g. mbed --profile mbed-os/tools/profiles/debug.json) to see if any RTX errors are printed that could shed light on why it doesn't work on that target?

jarlamsa, 2018-01-17:
Tried with the debug profile, not printing any errors.

samchuarm, 2018-01-19:
Hi @Archcady can you please help look at this Serial interface problem for RTL8195AM platform? Thanks.

prashantrar, 2018-02-01:
Just updating, I will be raising a PR for this issue tonight for review.

jarlamsa, 2018-02-23:
https://github.com/ARMmbed/mbed-os/pull/6009 fixed the issue, closing this.

closed 2018-02-23 by jarlamsa