ARMmbed/mbed-os #5810: NUCLEO-L476. How to use DMA.

lelely2, 2018-01-09:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug | Enhancement | Question
- Related issue: `#abc`
- Priority: Blocker | Major | Minor

---------------------------------------------------------------
## Bug

Target
 NUCLEO-L476

Toolchain:
MDK-ARM

Toolchain version:
5.22

mbed-cli version:

mbed-os sha:

DAPLink version:

Expected behavior
 Arm mbed-os should support NUCLEO-L476 DMA for UART.

Actual behavior
Using different OS works well with DMA for UART.

Steps to reproduce

0xc0170, 2018-01-09:
What's your use case for Serial ? To understand better the requirements 

cc @ARMmbed/team-st-mcd

bcostm, 2018-01-09:
DMA is not supported on mbed today. It is not related to ST devices... Can you please remove the ST label ? (for our analytics...)

0xc0170, 2018-01-09:
> Related #4360

Thanks @lelely2 for the request. I'll close this as not currently planned.

closed 2018-01-09 by 0xc0170