ARMmbed/mbed-os #5814: ImportError: No module named utils

SierpG, 2018-01-09:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
We are using the command "mbed compile ..." in Jenkins and get the following error message:

C:\Jenkins\workspace\Test2_Gabi>mbed compile -m ROBU_EU -t IAR --profile C:\Jenkins\workspace\M2M_Platform_ROBU_EU\mbed-os\tools\profiles\m2m_debug_test.json -vv 

Traceback (most recent call last):
  File "C:\Jenkins\workspace\Test2_Gabi\mbed-os\tools\make.py", line 51, in <module>
    from utils import argparse_filestring_type
ImportError: No module named utils

[mbed] Working path "C:\Jenkins\workspace\Test2_Gabi" (program)
[mbed] Exec "python -u C:\Jenkins\workspace\Test2_Gabi\mbed-os\tools\make.py -t IAR -m ROBU_EU --profile C:\Jenkins\workspace\M2M_Platform_ROBU_EU\mbed-os\tools\profiles\m2m_debug_test.json --source . --build .\BUILD\ROBU_EU\IAR -v" in C:\Jenkins\workspace\Test2_Gabi
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Jenkins\workspace\Test2_Gabi\mbed-os\tools\make.py -t IAR -m ROBU_EU --profile C:\Jenkins\workspace\M2M_Platform_ROBU_EU\mbed-os\tools\profiles\m2m_debug_test.json --source . --build .\BUILD\ROBU_EU\IAR -v" in "C:\Jenkins\workspace\Test2_Gabi"

If I look at the code in make.py it seems to me that in the lines 51 to 53 "from utils import ..." should be replaced by "from tools.utils import ..." as it is done in lines 32 and 33. As in line 30 the sys.path is extended by the root directory of the tools-directory utils.py should then be found.

0xc0170, 2018-01-09:
> If I look at the code in make.py it seems to me that in the lines 51 to 53 "from utils import ..." should be replaced by "from tools.utils import ..." as it is done in lines 32 and 33. As in line 30 the sys.path is extended by the root directory of the tools-directory utils.py should then be found.

Seems like that, does that fix the error you are having?

cc @theotherjimmy

theotherjimmy, 2018-01-09:
@SierpG That's correct. Would you like to submit that PR?

SierpG, 2018-01-11:
This fix solves our problem. However we can not yet submit a PR. We are working on this.

SierpG, 2018-01-11:
In the file tools/test.py the same problem exists in line 38 and 39. Here also "from utils import ..." should be replced by "from tools.utils import ...".