ARMmbed/mbed-os #5815: eventQueue Nrf52 Mbed Bug

ZakariaZed, 2018-01-09:
Hi everyone, 
I have a probleme in my code when i use the mbed-os-eddystone-service exemple. i modified the exemple to had mor services " TLM, UID, Battery" services and i added another frame " iBeacon frame". 
When i download the code in the device : 
* It work normaly. But when i turn off and turn it on it's make a problem.
So for the detection of problem i added a LED  in the fonctionand debug and i saw that the BUG is in the function : 
* eventQueue.call_every 
*  eventQueue.call_in
Please any suggestion.
Thank's.
--------------------------------------------------------------
Main Code : 

...
code...
........
static EventQueue eventQueue(/* event count */ 32 * EVENTS_EVENT_SIZE);

DigitalOut led(LED1, 1);
DigitalOut led2(LED2, 1);
DigitalOut led3(LED3, 1);
DigitalOut led4(LED4, 1);

/**
 * Callback triggered upon a disconnection event.
 */
static void disconnectionCallback(const Gap::DisconnectionCallbackParams_t *cbParams)
{
     uint8_t L=0;
     AuthentificationServicePtr->resetcode(L);
     
    (void) cbParams;
    BLE::Instance().gap().startAdvertising();
}

/**
 * Callback triggered some time after application started to switch to beacon mode.
 */

static void timeout(void)
{
     BLE &ble = BLE::Instance();
    Gap::GapState_t state;
    state = BLE::Instance().gap().getState();
    if (!state.connected) { /* don't switch if we're in a connected state. */
			led2 = !led2;
        eddyServicePtr->startBeaconService();
        AuthentificationServicePtr= new AuthentificationService(ble);
        batteryServicePtr = new BatteryService(ble, batteryLevel);
        eddyServicePtr->startConfigServiceafterdeconnexion();
        EddystoneService::EddystoneParams_t params;
        eddyServicePtr->getEddystoneParams(params);
        saveEddystoneServiceConfigParams(&params);
    } else {
			led3 = !led3;
        eventQueue.call_in(1000, timeout);
    }
}

..
.
.
code...
.
....

static void blinky(void)
{
    led = !led;
    BLE &ble = BLE::Instance();
/*
 * Appel de la fonction updatetempSensorValue chaque 500 ms.
 */ 
    
    if (ble.gap().getState().connected) {
            
            eventQueue.call(updatetempSensorValue);

    }
}

 static void onBleInitError(BLE::InitializationCompleteCallbackContext* initContext)
{
    /* Initialization error handling goes here... */
    (void) initContext;
}

static void initializeEddystoneToDefaults(BLE &ble)
{
.
.
.
code...
.
.
.
......
}

..
.
code...
....
static void bleInitComplete(BLE::InitializationCompleteCallbackContext* initContext)
{
    BLE         &ble  = initContext->ble;
    ble_error_t error = initContext->error;

    if (error != BLE_ERROR_NONE) {
        onBleInitError(initContext);
        return;
    }

    ble.gap().onDisconnection(disconnectionCallback);

    EddystoneService::EddystoneParams_t params;
  //  eddyServicePtr->setiBeacon(uuid, majorNumber, minorNumber ,txPower,companyID);  
    if (loadEddystoneServiceConfigParams(&params)) {
        AuthentificationServicePtr= new AuthentificationService(ble);
        eddyServicePtr = new EddystoneService(ble, params, radioPowerLevels, eventQueue);
        batteryServicePtr = new BatteryService(ble, batteryLevel); 
     } else {
        initializeEddystoneToDefaults(ble);
    }
led4 = !led4;
    /* Start Eddystone in config mode */
   eddyServicePtr->startConfigService();
/***************************************************************************************************/    
/*******************************THIS IS THE PROBLEM I GUESS***************************************/
/****************************************************************************************************/    
eventQueue.call_in( 1000, timeout);
/****************************************************************************************************/
/****************************************************************************************************/
/****************************************************************************************************/
}

void scheduleBleEventsProcessing(BLE::OnEventsToProcessCallbackContext* context) {
    BLE &ble = BLE::Instance();
    eventQueue.call(Callback<void()>(&ble, &BLE::processEvents));
}

int main()
{
    /* Tell standard C library to not allocate large buffers for these streams */
    setbuf(stdout, NULL);
    setbuf(stderr, NULL);
    setbuf(stdin, NULL);
/****************************************************************************************************/
/********************************THIS IS THE PROBLEM I GUESS************************************/
/****************************************************************************************************/
    eventQueue.call_every(500, blinky);
    eventQueue.call_every(70,updateSensorValue );
/****************************************************************************************************/
/****************************************************************************************************/
/****************************************************************************************************/    
    BLE &ble = BLE::Instance();
    ble.onEventsToProcess(scheduleBleEventsProcessing);
    ble.init(bleInitComplete);

    eventQueue.dispatch_forever();
    return 0;
}

0xc0170, 2018-01-09:
@pan- @paul-szczepanek-arm Please review

pan-, 2018-01-09:
@ZakariaZed Could you describe precisely what the issue is ? 
It would be also interesting to provide the mbed OS and compiler version used.

paul-szczepanek-arm, 2018-01-09:
is the code available to view on github?

ZakariaZed, 2018-01-09:
@pan-  I'm using Mbed OS 5 ans i use the online compiler and in the same time i tried with Uvision 5 " Keil" compiler (ARM Compiler : default compiler version 5).
I have a problem with eventQueue: 
when i dwonload the code in Nrf52832 it working normaly 
But if i turn off the device and turn on . it doesn't work anymore until I download the code again.
[file:///C:/Users/Algoria/Desktop/Eddystone%20final/26732750_1790802430978857_160679046_o.png](url)
[file:///C:/Users/Algoria/Desktop/Eddystone%20final/26735282_1790802470978853_1334892657_o.png](url)
[file:///C:/Users/Algoria/Desktop/Eddystone%20final/26735824_1790802520978848_1349672184_o.png](url)
[file:///C:/Users/Algoria/Desktop/Eddystone%20final/26771940_1790802444312189_907652736_o.png](url)
And when I turn it off and on it will be like : 
[file:///C:/Users/Algoria/Desktop/Eddystone%20final/26781500_1790802377645529_1802977526_o.png](url)

ZakariaZed, 2018-01-09:
@paul-szczepanek-arm  I'll Upload it

paul-szczepanek-arm, 2018-01-09:
@ZakariaZed thanks, can you please do it in at least two commits so that we can easily see your changes? thanks

ZakariaZed, 2018-01-09:
[https://github.com/ZakariaZed/mbed-os-example-ble-EddystoneService_URL-UID-TLM-iBeacon/blob/master/source/main.cpp](url)

@paul-szczepanek-arm 
@pan- 
This is my code and the problem was in the man.cpp.

ZakariaZed, 2018-01-09:
@pan- 
[https://github.com/ZakariaZed/mbed-os-example-ble-EddystoneService_URL-UID-TLM-iBeacon/tree/master/pictures](url)

pan-, 2018-01-09:
Looking at the code you shared with us, you use mbed OS 5.5.2. We solved many issues in the ticker and sleep behavior of the NRF52 with subsequent releases. I encourage you to try to reproduce the issue with mbed os 5.7.2 .

ZakariaZed, 2018-01-09:
@pan-  so the solution is to use Mbed os 5.7.2 ,

pan-, 2018-01-10:
@ZakariaZed maybe; try it and tell us.

ZakariaZed, 2018-01-10:
@pan-  Can you tel me how to do an update of Mbed os please ?

ZakariaZed, 2018-01-10:
@pan- @paul-szczepanek-arm  it working. Thank you :)

pan-, 2018-01-10:
Glad it help; I'm closing this issue.

closed 2018-01-10 by pan-