ARMmbed/mbed-os #5817: K60F support?

drahnr, 2018-01-09:
### Description
- Type: Question
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
K64F

## Question

To what extent are

K60F (this is the most important to me)
K63F
K65F
K81F

supported by the `K64F` target supported?

0xc0170, 2018-01-10:
@mmahadevan108 Please review

mmahadevan108, 2018-01-12:
The support should be fairly similar. However you might need to update the startup and linker files found under the device folder. You can download the files from https://mcuxpresso.nxp.com/en/select

drahnr, 2018-01-12:
Does this only apply to K60F or also to the other targets?

mmahadevan108, 2018-01-12:
Applies to all the targets mentioned in the list. Please use the mbed support for K64, K66 and K82 as reference and download the SoC specific files from  https://mcuxpresso.nxp.com/en/select

0xc0170, 2018-01-16:
@drahnr If that answers your question, please close it

closed 2018-01-16 by drahnr