ARMmbed/mbed-os #5820: LittleFS fopen in 'w' mode removes existing file

janjongboom, 2018-01-10:
From https://os.mbed.com/questions/79863/LittleFileSystem-fopenfilew-does-remove-/

> According to the C++ fopen documentation at http://www.cplusplus.com/reference/cstdio/fopen/ if 'mode' is "w" and the filename points to an existing file, the contents of the existing file are cleared.

> FATFileSystem does have this behaviour but LittleFileSystem does not seem to. Here's some example code....

```cpp
char * filename = "/fs/eventlog.bin";
FILE *fd = fopen(filename,"w");
fclose(fd);
struct stat stat_buf;
stat(filename, &stat_buf);
ASSERT(stat_buf.st_size!=0,"file size should be zero after opening");
```

> When the underlying filesystem is LittleFileSystem and if the file already exists and it not empty, this asserts since the filesize is not changed on fopen.

> Obviously a workaround is to use the remove method (or possibly a zero-length write?)

janjongboom, 2018-01-10:
@geky

geky, 2018-01-11:
Ah! This is the same as this issue, just hidden under a few layers: https://github.com/geky/littlefs/issues/7

Here's a patch that will fix this: https://github.com/ARMmbed/mbed-os/pull/5832

NeilMacMullen, 2018-01-18:
@geky   thanks - only just saw this.

closed 2018-01-18 by cmonr