ARMmbed/mbed-os #5835: Tickers update should be atomic

pan-, 2018-01-12:
### Description
- Type: Bug 

## Bug

The [function](https://github.com/ARMmbed/mbed-os/blob/master/hal/mbed_ticker_api.c#L111) updating the current ticker value does not do it atomically, as a result concurrent updates of the ticker may corrupt the ticker value. This may happen if multiples thread read concurrently the ticker value.

Protecting the data structure update from concurrent updates with a critical section may be a solution.

0xc0170, 2018-01-12:
> This may happen if multiples thread read concurrently the ticker value.

👍 

Looking at the file who handles the critical section, caller does in most cases, invoking update time within critical section. The only case it fails is when reading the time as reported in `ticker_read_us` (I do not count initialize as that one is run once and at the end is event scheduled).

@c1728p9 @SenRamakri @bulislaw  please review

NeilMacMullen, 2018-01-12:
FWIW, I believe this matches the sc

mprse, 2018-01-19:
Fix for this issue has been provided here: PR https://github.com/ARMmbed/mbed-os/pull/5889.

NeilMacMullen, 2018-01-19:
I've just patched this across to my build and initial impressions are that it has fixed the issue.  Thanks!

cmonr, 2018-02-02:
Closing since all PRs appear to be merged and issue appears to be fixed.

closed 2018-02-02 by cmonr