ARMmbed/mbed-os #5838: ublox NINA B112 does NOT reset over serial.

ashok-rao, 2018-01-12:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
EVK-NINA-B112

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
a5f1426

**DAPLink version:**
0244

**Expected behavior**
Sending "break" over serial should reset the MCU.

**Actual behavior**
Sending "break" over serial does NOT reset the MCU.

**Steps to reproduce**
1. Build any application such as blinky with a Serial object.
2. Include a printf to obtain messages on the serial console. Preferably just once upon reset.
3. Build image with above toolchains, flash the target via any DAPLink method.
4. Run the application.
5. We see that the device needs to be power cycled to run the application and also sending "break" from PuTTY does not reset the MCU. Expected behaviour is to print the message every time the MCU is reset.

Also note HW flow control is disabled for the target in mbed_app.json. With FC enabled, the application does not run at all.

@andreaslarssonublox  @c1728p9  @0xc0170 , could you please check? Thanks.

0xc0170, 2018-01-12:
@ARMmbed/team-ublox

andreaslarssonublox, 2018-01-15:
Hi, 
Some questions:
What debug chip is used?
Schematics?
What is the target here, EVK-NINA-B112(#5838) or MTB_UBLOX_NINA_B1(#5837)?

ashok-rao, 2018-01-15:
Thanks @andreaslarssonublox . Debug chip = LPC11U35 . I'm not sure if I can share the schematics at the moment. Let me check that with @chris-styles. Target = MTB_UBLOX_NINA_B1 but it is using the NINA-B112 module (same as the EVK-NINA-B112) and brings out the Tx, Rx, NRESET pins from the module to the breakout board that are mapped in PinNames.h as part for PR #5837 ..

ashok-rao, 2018-01-17:
@andreaslarssonublox .. bump!

andreaslarssonublox, 2018-01-18:
Hi, I looked at https://github.com/ARMmbed/DAPLink but could not find any support for the combination LPC11U35 and NINA-B1 and we don't have any hardware to test with so might be hard for us to debug?

ashok-rao, 2018-01-26:
This is DAPLink fix. New version of DAPLink here: https://github.com/ARMmbed/DAPLink/pull/376 fixes this issue.
Closing. Thanks!

closed 2018-01-26 by ashok-rao