ARMmbed/mbed-os #5839: NUCLEO-F429ZI export to uVision 5 is broken (online compiler)

janjongboom, 2018-01-12:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
NUCLEO-F429ZI

**Toolchain:**
ARM

**Toolchain version:**
Online Compiler

**Expected behavior**
Export succeeds

**Actual behavior**
Export fails

**Steps to reproduce**
Apparently does blinky work, but HelloMQTT does not...

See https://os.mbed.com/questions/79879/Unable-to-export-helloMQTT-and-CoAP-exam/

0xc0170, 2018-01-12:
@theotherjimmy Please review

cmonr, 2018-01-17:
Reproduced with the following commands:
```mbed import https://os.mbed.com/teams/mqtt/code/HelloMQTT/ HelloMQTT
cd HelloMQTT
mbed export -m NUCLEO_F429ZI -i uvision5```

Gave the following output:
```
[mbed] WARNING: Could not find mbed program in current path "/home/mrcruz/Work".
[mbed] WARNING: You can fix this by calling "mbed new ." in the root of your program.
---
Scan: .
Warning: Multiple linker scripts detected: ./ARM-mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct -> ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct
Warning: Multiple linker scripts detected: ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct -> ./mbed-os-example-blinky/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct
Warning: Multiple linker scripts detected: ./mbed-os-example-blinky/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct -> ./mbed-os-issues/5348/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct
Warning: Multiple linker scripts detected: ./mbed-os-issues/5348/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct -> ./mbed-os-issues/5676/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct
Warning: Multiple linker scripts detected: ./mbed-os-issues/5676/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct -> ./mbed-os-issues/5706/mbed-os-example-tls/authcrypt/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct
Warning: Multiple linker scripts detected: ./mbed-os-issues/5706/mbed-os-example-tls/authcrypt/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct -> ./mbed-os-issues/5706/mbed-os-example-tls/benchmark/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct
Warning: Multiple linker scripts detected: ./mbed-os-issues/5706/mbed-os-example-tls/benchmark/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct -> ./mbed-os-issues/5706/mbed-os-example-tls/hashing/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct
Warning: Multiple linker scripts detected: ./mbed-os-issues/5706/mbed-os-example-tls/hashing/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct -> ./mbed-os-issues/5706/mbed-os-example-tls/tls-client/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct
Warning: Multiple linker scripts detected: ./mbed-os-issues/5706/mbed-os-example-tls/tls-client/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct -> ./OOB/mbed-os-example-blinky/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct
Warning: Multiple linker scripts detected: ./OOB/mbed-os-example-blinky/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct -> ./OOB/mbed-os-example-filesystem/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct
Warning: Multiple linker scripts detected: ./OOB/mbed-os-example-filesystem/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct -> ./tmp/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct
Warning: Multiple linker scripts detected: ./tmp/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct -> ./tools/ARM-mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct
Warning: Multiple linker scripts detected: ./tools/ARM-mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct -> ./tools/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct
Warning: Multiple linker scripts detected: ./tools/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct -> ./tools/test/Before/mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_ARM_STD/stm32f429xx.sct
Traceback (most recent call last):
  File "/home/mrcruz/Work/mbed-os/tools/project.py", line 259, in <module>
    main()
  File "/home/mrcruz/Work/mbed-os/tools/project.py", line 254, in main
    build_profile=profile, app_config=options.app_config)
  File "/home/mrcruz/Work/mbed-os/tools/project.py", line 96, in export
    app_config=app_config)
  File "/home/mrcruz/Work/mbed-os/tools/export/__init__.py", line 319, in export_project
    for loc, path in src_paths.iteritems()}
  File "/home/mrcruz/Work/mbed-os/tools/export/__init__.py", line 319, in <dictcomp>
    for loc, path in src_paths.iteritems()}
  File "/home/mrcruz/Work/mbed-os/tools/build_api.py", line 436, in scan_resources
    resources = toolchain.config.load_resources(resources)
  File "/home/mrcruz/Work/mbed-os/tools/config/__init__.py", line 893, in load_resources
[... cut ...]
tools.config.ConfigException: Library name 'events' is not unique (defined in '/home/mrcruz/Work/HelloMQTT/mbed-os/events/mbed_lib.json' and '/home/mrcruz/Work/ARM-mbed-os/events/mbed_lib.json')
[mbed] ERROR: "/home/mrcruz/.local/share/virtualenvs/ARM-py2-env-vzLJPstb/bin/python2" returned error code 1.
[mbed] ERROR: Command "/home/mrcruz/.local/share/virtualenvs/ARM-py2-env-vzLJPstb/bin/python2 -u /home/mrcruz/Work/mbed-os/tools/project.py -i uvision5 -m NUCLEO_F429ZI --source ." in "/home/mrcruz/Work"
---
```

cmonr, 2018-01-17:
On the other hand, this was/is reproducible with the online IDE. @0xc0170, don't think @theotherjimmy is needed for this one.

jeromecoutant, 2018-10-02:
@adbridge  Please re-open

jarlamsa, 2018-10-29:
Is this being worked on? Seems that it is not working with offline exporter as well. You can run the export command and it creates the needed files, but when you try opening the project, uVision asks you to install the needed device family pack, after it is installed, uVision says: 
Error: Device not found -
Device: 'STM32F429ZI'
Vendor: 'STMicroelectronics'
Please update your device selection

Steps to reproduce:
```
mbed import mbed-os-example-blinky
cd mbed-os-example-blinky
mbed export -m NUCLEO_F429ZI -i uvision5
```
Open the project with uvision5

0xc0170, 2018-10-29:
@jarlamsa We experienced similar issue in the CI recently. What is your uvision version? If you use older uvision 5.24 or similar (I installed latest 5.26 from September 2018), it can't recognize this older device name and set it to the new one, so rather it prints an error. The newer version fixes it and automatically resolves this.

0xc0170, 2018-10-29:
> @adbridge Please re-open

@jeromecoutant What is the current state of this issue ? Is there still bug in the online enviroment or somewhere else?

jarlamsa, 2018-10-29:
@0xc0170 I have 5.24.2.0, good information, I'll update and try again, thanks.

janjongboom, 2018-10-29:
@jarlamsa Yes, with latest version of uVision this worked for me on the F429ZI (different project).

jeromecoutant, 2018-11-05:
> @jeromecoutant What is the current state of this issue ? Is there still bug in the online enviroment or somewhere else?

I didn't check. I requested Anna to open again the issue, as issue was opened by @janjongboom 
So for me, it has to be closed by him, not a ghost script.

@janjongboom 
If it works on your side, maybe you can close the issue ?
Thx

closed 2018-11-05 by theotherjimmy