ARMmbed/mbed-os #5843: Nordic RF52 Separate TWI blocks cannot be used in parallel from different threads

NeilMacMullen, 2018-01-12:
Using RF52838, Any Mbed > 4.5

I'm not sure if this is a known or deliberate (but undocumented) restriction?

It seems not to be possible to use  multiple TWI blocks from different threads.  In practice I have seen this on a platform with both SPI (for file system) and I2C (for sensors/comms)

If the I2C bus is used on a background thread (to poll a sensor for example) and the file system is read (over SPI) the file system will fail to return to the caller (I suspect that the low level SPI code is returning a fault condition but haven't traced it this far).

NeilMacMullen, 2018-01-12:
Adding @MarceloSalazar and @dlfryar as watchers

screamerbg, 2018-01-30:
cc @nvlsianpu

marcuschangarm, 2018-03-08:
This should be fixed in this PR to our new feature branch: https://github.com/ARMmbed/mbed-os/pull/6296

Progress on the new feature branch can be followed here: https://os.mbed.com/forum/upcoming-features/topic/29477/

0xc0170, 2018-05-18:
The feature branch was integrated, this issue should be resolved

closed 2018-05-18 by 0xc0170