ARMmbed/mbed-os #5849: Erratic Thread::wait timing

scttnlsn, 2018-01-14:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
STM32F072RB

**Toolchain:**
GCC_ARM via PlatformIO

**Toolchain version:**
arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 6.2.1 20161205 (release) [ARM/embedded-6-branch revision 243739]

**Expected behavior**
1 second between LED blinks

**Actual behavior**
4 blinks at ~1Hz followed by 4 blinks at ~3Hz (estimated visually), cycle repeats

**Steps to reproduce**

Run the following code:

```c++
#include <mbed.h>
#include <rtos.h>

DigitalOut led(LED1);
Thread blinker;

void blinker_thread() {
  while (true) {
    led = !led;
    Thread::wait(1000);
  }
}

int main() {
  blinker.start(callback(blinker_thread));

  while (true) {
    Thread::wait(1000);
  }

  return 0;
}

```

chrissnow, 2018-01-14:
That sounds more like what happens when mbed crashes. 

Have you checked the serial debug for messages?

scttnlsn, 2018-01-14:
That sounds very likely to be the case.  I don't see anything on the USB serial (tried various baud rates).  How do I see the debug messages?  Thanks!

RaEyE, 2018-01-15:
The default Baudrate for most devices would be 9600Bauds 8Data 1Stop and no Parity.
In case you didn't configure something different (e.g. in code or with an override in mbed_app.json)

pan-, 2018-01-15:
Looks like you are running out of memory: The ` NUCLEO_F072RB` is limited to 16KB of RAM. The application occupies *statically* `10104` bytes of RAM which leaves ~6K of RAM for the application. 

When a thread is started, a chunk of 4096 bytes of memory is allocated and used as the thread stack. That leaves less than 2k of memory for the ISR stack and C/C++ runtime (registration of objects to delete at termination, I/Os, heap management, ...). 

To reduce the stack size used by the main thread you can override the definition `MBED_CONF_APP_MAIN_STACK_SIZE`. Similarly it is possible to reduce the thread stack size allocated when a thread is started by overriding the definition `MBED_CONF_APP_THREAD_STACK_SIZE`. You can try it on the command line: 

```
mbed compile -t GCC_ARM -m NUCLEO_F072RB -DMBED_CONF_APP_MAIN_STACK_SIZE=2048 -DMBED_CONF_APP_THREAD_STACK_SIZE=2048
```

This command will save 4096 bytes of memory by dividing the stack of the main thread and the stack of the blinker thread by 2.

---------

More generally using threading facilities may be challenging on devices with less than 64K of RAM. The [event queue](https://os.mbed.com/docs/v5.7/tutorials/the-eventqueue-api.html) can be used instead to schedule tasks.

scttnlsn, 2018-01-15:
Thanks, that was indeed the issue.  Should I have seen some output on the serial port when mbed runs out of memory?

closed 2018-01-15 by scttnlsn

pan-, 2018-01-15:
> Thanks, that was indeed the issue. Should I have seen some output on the serial port when mbed runs out of memory?

If the operator `new` fail to provide memory then the system enters in error mode and display a message.