ARMmbed/mbed-os #5850: NUCLEO_L476RG: LowPowerTicker::atach () can only be specified for a short time

maskedw, 2018-01-15:
### Description
- Type: Bug

In `mbed_ticker_api.c`, management is performed considering the maximum value of the hardware timer, but` mbed_lp_tiker_api.c` seems not to handle the value of `us_timesamp_t (uint64_t)` properly.

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_L476RG

**Toolchain:**
GCC_ARM

**Toolchain version:**
```sh
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
```

**mbed-cli version:** 
```sh
$ mbed --version
1.2.2
```
**mbed-os sha:**
```sh
$ git log -n1 --oneline
eca67ca (grafted, HEAD, tag: mbed-os-5.7.2, tag: latest) Merge pull request #5767 from ARMmbed/release-candidate
```
**Steps to reproduce**
```c++
#include "mbed.h"

DigitalOut led1(LED1);
Serial pc(USBTX, USBRX);
static void NullFunc() {}

int main() {
    pc.baud(115200);
    
        int times[] = {
        10,
        60 * 60 * 3,
        60 * 60 * 17,
        60 * 60 * 25,
    };

    
    LowPowerTicker lpTicker;
    LowPowerTimer timer;
    timer.start();

    for (int i = 0; i < (int)sizeof(times) / sizeof(times[0]); i++)
    {
        const int t = times[i];
        printf("deepsleep    %10d[Sec]\n", t);
        const int start = timer.read_ms();
        lpTicker.attach(NullFunc, t);
        hal_deepsleep();
        lpTicker.detach();
        const int delta = timer.read_ms() - start;
        printf("Wakeup after %10d[MSec]\n", delta);
    }
    
    while (true) {
        led1 = !led1;
        wait(0.5);
    }
    return 0;
}

```
**Result**
```
[2018-01-13 19:04:32.542] deepsleep            10[Sec]
[2018-01-13 19:04:42.558] Wakeup after      10007[MSec]
[2018-01-13 19:04:42.558] deepsleep         10800[Sec]
[2018-01-13 19:36:02.002] Wakeup after    6174399[MSec]
[2018-01-13 19:36:02.002] deepsleep         61200[Sec]
[2018-01-13 20:07:22.025] Wakeup after    6174962[MSec]
[2018-01-13 20:07:22.025] deepsleep         90000[Sec]
[2018-01-13 20:38:42.036] Wakeup after    6174963[MSec]
```

0xc0170, 2018-01-16:
> arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

Not related, but use version 6

Are you using mbed OS 5 or mbed 2 with rtos?

> In mbed_ticker_api.c, management is performed considering the maximum value of the hardware timer, butmbed_lp_tiker_api.c seems not to handle the value of us_timesamp_t (uint64_t) properly.

Can you provide more details? Looks like you already looked at the code closer?

@ARMmbed/team-st-mcd

bcostm, 2018-01-16:
I checked this nice example on mbed IDE using mbed library and the result is completely incorrect.

Then I used mbed-os library instead (version 5.7.3) and result seems ok (but I didn't wait hours...):

```
deepsleep            10[Sec]
Wakeup after      10007[MSec]
deepsleep            60[Sec]
Wakeup after      59979[MSec]
deepsleep            70[Sec]
Wakeup after      69995[MSec]
```

maskedw, 2018-01-16:
> Are you using mbed OS 5 or mbed 2 with rtos?

I am using mbed OS5.

> Can you provide more details? Looks like you already looked at the code closer?

I'm sorry. There seemed to be a suspicious part, but it seems it was a misunderstanding.

If the value exceeds 32 bits per us, will it be divided into multiple interrupts?
Then, NullFunc () of the sample code has not been called.
I wanted to do more than 20 hours of deepsleep, but have to accept what is woken up on the way many times?

0xc0170, 2018-01-16:
Lets consider mbed OS 5 -  do not call deepsleep (not even hal deepsleep) directly if you are not having specific reason to do so. Let threads to become idle if there is nothing to do. 
If your target supports tickless mode, it would go to sleep automatically using low power timer (otherwise high freq ticker is used as rtos system timer, often it is systick). There is a limitation (how long a target can sleep - 32bit value that is in milliseconds - default rtos tick is 1ms). An application would go to sleep again if there is nothing to do.

> I checked this nice example on mbed IDE using mbed library and the result is completely incorrect.

Please keep us updated of your finding, thanks for testing !

bcostm, 2018-01-16:
1 hour delay: The time displayed is ms is wrong but the delay last effectively around 1 hour (I didn't measured it precisely but it was more than 1879s/60=31min).

```
Deepsleep          3600[Sec]
Wakeup after    1879052[MSec]
```

Now I'm testing 31min and 33min. I think there is a limitation to 32min...

maskedw, 2018-01-17:
```c++
#include "mbed.h"

DigitalOut led1(LED1);
Serial pc(USBTX, USBRX);
EventFlags event;

static void IdleHook()
{
    {
        CriticalSectionLock lock;
        sleep();
    }
}

static void Wakeup()
{
    event.set(1);
}

int main()
{
    pc.baud(115200);

    int times[] = {
        10,
        60 * 60 * 3,
        60 * 60 * 17,
        60 * 60 * 25,
    };

    Thread::attach_idle_hook(IdleHook);
    LowPowerTicker lpTicker;
    for (int i = 0; i < (int)sizeof(times) / sizeof(times[0]); i++)
    {
        const int t = times[i];
        printf("deepsleep    %10d[Sec]\n", t);
        const time_t start = time(NULL);
        lpTicker.attach(Wakeup, t);
        event.wait_all(1);
        lpTicker.detach();
        const time_t delta = time(NULL) - start;
        printf("Wakeup after %10d[Sec]\n", delta);
    }

    while (true) {
        led1 = !led1;
        wait(0.5);
    }
    return 0;
}
```

I changed the test code.

Since NUCLEO_L476RG does not seem to support TICKLESS mode, I set a custom idle hook.
I will report it when a result comes out.

maskedw, 2018-01-22:
It worked as expected.
```
[2018-01-17 19:15:34.137] deepsleep            10[Sec]
[2018-01-17 19:15:44.139] Wakeup after         10[Sec]
[2018-01-17 19:15:44.154] deepsleep            30[Sec]
[2018-01-17 19:16:14.173] Wakeup after         30[Sec]
[2018-01-17 19:16:14.173] deepsleep            60[Sec]
[2018-01-17 19:17:14.179] Wakeup after         60[Sec]
[2018-01-17 19:17:14.179] deepsleep          1800[Sec]
[2018-01-17 19:47:14.787] Wakeup after       1801[Sec]
[2018-01-17 19:47:14.787] deepsleep          3600[Sec]
[2018-01-17 20:47:15.818] Wakeup after       3601[Sec]
[2018-01-17 20:47:15.818] deepsleep         90000[Sec]
[2018-01-18 21:47:17.340] Wakeup after      90001[Sec]
[2018-01-18 21:47:17.356] deepsleep         90000[Sec]
[2018-01-19 22:47:21.210] Wakeup after      90001[Sec]
```

maskedw, 2018-01-22:
Sorry. This was not a bug.
I was misusing myself.

@0xc0170
@bcostm
Thank you for your help.

closed 2018-01-22 by maskedw