ARMmbed/mbed-os #5853: M487: mbed-os-tests-mbedtls-selftest test failed

ccli8, 2018-01-15:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NUMAKER_PFM_M487

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
`b590051 Merge pull request #5808 from SenRamakri/sen_ESP8266TestConfig`

**Greentea test report**
```
mbedgt: test case report:
+--------------------------+------------------+--------------------------------+---------------------------+--------+--------+--------+--------------------+
| target                   | platform_name    | test suite                     | test case                 | passed | failed | result | elapsed_time (sec) |
+--------------------------+------------------+--------------------------------+---------------------------+--------+--------+--------+--------------------+
| NUMAKER_PFM_M487-GCC_ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbedtls-selftest | mbedtls_entropy_self_test | 0      | 1      | FAIL   | 0.08               |
| NUMAKER_PFM_M487-GCC_ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbedtls-selftest | mbedtls_sha256_self_test  | 1      | 0      | OK     | 0.93               |
| NUMAKER_PFM_M487-GCC_ARM | NUMAKER_PFM_M487 | mbed-os-tests-mbedtls-selftest | mbedtls_sha512_self_test  | 1      | 0      | OK     | 0.62               |
+--------------------------+------------------+--------------------------------+---------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 1 FAIL / 2 OK
```

Per my check, there is a call flow error with SHA in `mbed-os/features/mbedtls/src/entropy.c`.
`mbedtls_sha256_init`/`mbedtls_sha512_init` is missing in `mbedtls_entropy_init`. If I fix it as below, the test would be OK:
<pre>
void mbedtls_entropy_init( mbedtls_entropy_context *ctx )
{
    memset( ctx, 0, sizeof(mbedtls_entropy_context) );

#if defined(MBEDTLS_THREADING_C)
    mbedtls_mutex_init( &ctx->mutex );
#endif

#if defined(MBEDTLS_ENTROPY_SHA512_ACCUMULATOR)
    <b>mbedtls_sha512_init( &ctx->accumulator );</b>
    mbedtls_sha512_starts( &ctx->accumulator, 0 );
#else
     <b>mbedtls_sha256_init( &ctx->accumulator );</b>
    mbedtls_sha256_starts( &ctx->accumulator, 0 );
#endif
</pre>

<pre>
void mbedtls_entropy_free( mbedtls_entropy_context *ctx )
{
#if defined(MBEDTLS_HAVEGE_C)
    mbedtls_havege_free( &ctx->havege_data );
#endif
<b>
#if defined(MBEDTLS_ENTROPY_SHA512_ACCUMULATOR)
    mbedtls_sha512_free( &ctx->accumulator );
#else
    mbedtls_sha256_free( &ctx->accumulator );
#endif
</b>
#if defined(MBEDTLS_THREADING_C)
    mbedtls_mutex_free( &ctx->mutex );
#endif
</pre>

0xc0170, 2018-01-15:
@hanno-arm @mazimkhan Please review

hanno-arm, 2018-01-15:
Thanks @ccli8 for reporting this. The code-paths you're referring to indeed only work if zeroization is proper initialization / freeing for SHA contexts, which for some alternative implementations is not the case. I will check if any work has already been done in this direction; if not, we should open a PR for the main Mbed TLS repository with your suggested fix.

hanno-arm, 2018-01-17:
@ccli8 I couldn't find any pre-existing work on the issue and opened PR's with your suggested fix in the main Mbed TLS repository. See https://github.com/ARMmbed/mbedtls/pull/1275. Thanks again for spotting and reporting this! :+1:

0xc0170, 2018-01-18:
@hanno-arm I see you did some extra bugfixing the PR. Shall this be closed once your fix is integrated and will be part of the next mbed tls version update for mbed OS?

hanno-arm, 2018-01-18:
@0xc0170 I have just raised this with @Patater who is planning the next release; it's due very soon, though, and I cannot guarantee the PR will make it.

hanno-arm, 2018-01-18:
@ccli8 How would you like to be credited in the ChangeLog?

ccli8, 2018-01-19:
> How would you like to be credited in the ChangeLog?

@hanno-arm sorry, what do you mean?

hanno-arm, 2018-01-19:
@ccli8 In https://github.com/ARMmbed/mbedtls/pull/1275 we mention that you have reported the bug that the PR sets out to fix, through an entry in the [ChangeLog](https://github.com/ARMmbed/mbedtls/pull/1275/files#diff-02f0b547c2779d25cff89672135f20e3R60) of Mbed TLS. Could you let us know whether (a) You are ok with that, and (b) If yes, if you would like to be credited under your GitHub name (as of now) or, e.g., your full name and affiliation?

ccli8, 2018-01-19:
@hanno-arm The change is OK. Just use my GitHub name (ccli8) for the change log.

hanno-arm, 2018-01-19:
@ccli8 Ok, thanks for confirming.

ccli8, 2018-03-12:
Closed with https://github.com/ARMmbed/mbedtls/pull/1275 merged.

closed 2018-03-12 by ccli8