ARMmbed/mbed-os #5859: NUCLEO_L476RG, IAR does not link

teetak01, 2018-01-16:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_L476RG

**Toolchain:**
|IAR

**Toolchain version:**
   IAR ANSI C/C++ Compiler V7.80.1.11864/W32 for ARM

**mbed-os sha:**
Mbed OS 5.7.2/5.7.3

**Expected behavior**
Application links.

**Actual behavior**
```
Link: mbed-os-example-client
Error[Lp011]: section placement failed
            unable to allocate space for sections/blocks with a total estimated
                      minimum size of 0x19834 bytes (max align 0x8) in
                      <[0x20000000-0x20017fff]> (total uncommitted space
                      0x18000).
[ERROR] Error[Lp011]: section placement failed
            unable to allocate space for sections/blocks with a total estimated
                      minimum size of 0x19834 bytes (max align 0x8) in
                      <[0x20000000-0x20017fff]> (total uncommitted space
                      0x18000).
```
**Steps to reproduce**

Use wifi_idw01m1_v4.json configuration for mbed-os-example-client:

Compile with:
`mbed compile  -t IAR -m NUCLEO_L476RG
`

0xc0170, 2018-01-16:
What version did it link before to be able to compare? The error is regarding RAM area - not enough space left for the app to fit. I can see this was tested using example client

The numbers specify SRAM section

```
define symbol __region_SRAM1_start__  = 0x20000000;
define symbol __region_SRAM1_end__    = 0x20017FFF;
```

This might be similar patch that could help: https://github.com/ARMmbed/mbed-os/pull/5844 

@ARMmbed/team-st-mcd

teetak01, 2018-01-16:
Not tested with any older versions.

adustm, 2018-01-16:
Hello @teetak01 
I have faced the same kind of link problem for the DISCO_L475, that 's why I've created #5844. 
You can apply the same modification on 
mbed-os/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L476xG/device/TOOLCHAIN_IAR/stm32l476xx.icf
```
@@ -21,12 +21,11 @@ define region ROM_region = mem:[from __region_ROM_start__ to __region_ROM_end__]
 define region SRAM2_region = mem:[from __region_SRAM2_start__ to __region_SRAM2_end__];
 define region SRAM1_region = mem:[from __region_SRAM1_start__ to __region_SRAM1_end__];
-/* Stack 1/8 and Heap 1/4 of RAM */
-define symbol __size_cstack__ = 0x8000;
-define symbol __size_heap__   = 0xa000;
+/* Stack complete SRAM2 and Heap 2/3 of SRAM1 */
+define symbol __size_cstack__ = 0x7e00;
+define symbol __size_heap__   = 0x10000;
 define block CSTACK    with alignment = 8, size = __size_cstack__   { };
 define block HEAP      with alignment = 8, size = __size_heap__     { };
-define block STACKHEAP with fixed order { block HEAP, block CSTACK };
@@ -34,5 +33,5 @@ do not initialize  { section .noinit };
 place at address mem:__intvec_start__ { readonly section .intvec };

 place in ROM_region   { readonly };
-place in SRAM1_region   { readwrite, block STACKHEAP };
-place in SRAM2_region { };
+place in SRAM1_region { readwrite, block HEAP };
+place in SRAM2_region { block CSTACK };

```

Let us know if it fixes your issue and I will deploy the PR to the STM32L476 devices.

bcostm, 2018-02-02:
The PR #6004 should solve this issue.

closed 2018-02-07 by cmonr