ARMmbed/mbed-os #5860: Data Storage Mbed

ZakariaZed, 2018-01-16:
Hi,

   I'm using Nrf52 with Mbed ble exemple and i want to save data when i try to modify caracteristics. The problem is when i turn off my device and turn it on, it take the old value not my value. 
So i want to know what i have to modify for saving value.
Thank's,
Zakaria.

0xc0170, 2018-01-16:
@ZakariaZed Can you share any code, what version of mbed-os, which ble example, anything that can help a reader ?

ZakariaZed, 2018-01-16:
@0xc0170 

 So i use the last version of Mbed-os, i think is 5.7.2. " I just update Mbed-os at the online mbed compiler. 
The exemple is : "Mbed-os-EddystoneService" that i modified to send more data.
When i try to modify period caracteristics for exemple, the value are changed. When i disconnect and connet to the device, the value still there BUT when i turn off the device and turn On. the value take the default value not mine.

0xc0170, 2018-01-16:
@pan-

ZakariaZed, 2018-01-16:
Hi,

    I think i have a problem with saveEddystoneServiceConfigParams(&params) function.
Now when i try to modify a value. and do the samethink before ( i mean turn off and on). it make me random variable.
For exemple, i change period from 700 ms to 722ms. after it display 704 ms !!!!!!!!

ZakariaZed, 2018-01-17:
Hi,

I think that when i want to startbeacon service function with creating services and connecting every time i want to the device.

This is why i got i problem with "saveEddystoneServiceConfigParams" to save params after modification.

If i add to the function timeout a setupconfig to creat service. i can't save params.

Can someone tel me why.
Thank's
Zakaria.

pan-, 2018-01-17:
@ZakariaZed IIRC storage only works on NRF51. The Eddystone service present in BLE is also completely outdated and won't be updated in the near future. In the meantime there is an mbed implementation maintained by google: https://github.com/google/eddystone/tree/master/implementations/mbed ; please use this one.

ZakariaZed, 2018-01-18:
The problem Was with the saveEddystoneServiceConfigParams function : 
It was like : 
void saveEddystoneServiceConfigParams(const EddystoneService::EddystoneParams_t *paramsP)
{
    memcpy(&persistentParams.params, paramsP, sizeof(EddystoneService::EddystoneParams_t));
    if (persistentParams.persistenceSignature != PersistentParams_t::MAGIC) {
        persistentParams.persistenceSignature = PersistentParams_t::MAGIC;
    } else {
        fs_erase(&fs_config, fs_config.p_start_addr, sizeof(PersistentParams_t) / 4);
    }

    fs_store(&fs_config,
             fs_config.p_start_addr,
             reinterpret_cast<uint32_t *>(&persistentParams),
             sizeof(PersistentParams_t) / 4);
}

and i change it to : 
void saveEddystoneServiceConfigParams(const EddystoneService::EddystoneParams_t *paramsP)
{
    memcpy(&persistentParams.params, paramsP, sizeof(EddystoneService::EddystoneParams_t));
    if (persistentParams.persistenceSignature == PersistentParams_t::MAGIC) {
       // printf("persistentParams.persistenceSignature == PersistentParams_t::MAGIC\n");
        fs_erase(&fs_config, fs_config.p_start_addr, 1);
        
    }
    //printf("aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa\n");
    persistentParams.persistenceSignature = PersistentParams_t::MAGIC;

    fs_store(&fs_config,
             fs_config.p_start_addr,
             reinterpret_cast<uint32_t *>(&persistentParams),
             sizeof(PersistentParams_t) / 4);
}

PS : the problem was in the MAGIC number.

donatieng, 2018-07-02:
Closing as resolved.

closed 2018-07-02 by donatieng