ARMmbed/mbed-os #5863: FlashIAP API should allow smaller program sizes then sector size

helmut64, 2018-01-17:
- Type: Enhancement

Usually when a flash chip is erased (e.g. 2kB sector size) it is possible to write multiple times to this flash sector., e.g. at offset 0, 8 bytes, and later at offset 8 another 8 bytes at another time. To allow this the FlashIAP API and implementation needs to be extended, API e.g.:

int minProgramSize(); // gets the minium program size often 8 bytes.
int programAlignment(); // gets the programm alignment often 64-bit

This would allow more effective handling of flash programming, it would also be helpful for littlefs block device drivers.

0xc0170, 2018-01-17:
What would be difference between program page size and min program page size. Would ti be that program page size returns the ideal size for most use cases and minimum the minimum size that can be programmed. In some cases this could be as you defined 8 bytes the minimum, page size would be 1k ?

> programAlignment

how to use this one?

> This would allow more effective handling of flash programming, it would also be helpful for littlefs block device drivers.

Can you give examples to illustrate

@c1728p9 @SenRamakri

helmut64, 2018-01-17:
The max Program page size depends on the chip, usually it is 256 bytes or even up to the sector size (e.g. 2kB). This depends on the flash chip. A flash erase sets the entire sektor  to 0xff's (2kB in my STM case). The erase is the costly operation because you have a chip depending max. erase cycle, it is somewhere between 1000 and 100k times erase, depending on the type/vendor. Wringing (program) can can done multiple times, however the write will only set high bits to low, the same block allows multiple writes, however only changing bits from high to low, otherwise an erase before is required.

When the API allows smaller program sizes (as supported by the chips) we need only a fraction of the erases. That is the idea.

I can create a patch, however every MCU vendor needs to do the HAL backend adjustment, therefore I believe it is better in ARM hands.

saedelman, 2018-02-07:
I was just going to request an enhancement on the FlashIAP also when I found this thread. Good example is the nRF52 series chip. The minimum program size is 4 bytes but the FlashIAP library insists on writing only pages of 4K. This amount may be readily allocatable statically or dynamically in a bootloader, but if you are trying to update the bootloader itself from the main application, you may not have 4K laying around to be able to allocate. Right now, I'm commenting out the following in the program() function in FlashIAP.cpp to get around it:

` if (!is_aligned(addr, page_size) ||
        !is_aligned(size, page_size) ||
        (size < page_size) ||
        (((addr % current_sector_size) + size) > current_sector_size)) {
        return -1;
    }
`

saedelman, 2018-02-07:
Turns out that the all flash erase/write operations used in the FlashIAP library fail when the SoftDevice (sd) in the nRF52 is enabled. Looking at the Nordic HAL driver, they are using NVMC calls always when sd calls should be made when the sd is enabled. It would be nice if the FlashIAP library could work on nRF52 whether the softdevice is enabled or not. The Nordic "fstorage" library included in mBED appears to be able to exactly that, not sure why the Nordic HAL driver does not sit on top of fstorage?

helmut64, 2018-02-08:
@saedelman the SD on nRF52 is an nRF52 HAL specific problem, so a different case.
My attempt is to make sure that the FlashIAP API is flexible to allow what today MCUs internal flash can handle. I don't know any MCU which requires to programm entire flash selctors at once, they all allow to write in smaller chunks, they allow to write multiple times to the same flash sector (of course into regions/offsets which are still erased (0xff's)

0xc0170, 2018-02-08:
> Turns out that the all flash erase/write operations used in the FlashIAP library fail when the SoftDevice (sd) in the nRF52 is enabled. Looking at the Nordic HAL driver, they are using NVMC calls [...]

Please file a new issue for this

0xc0170, 2018-02-08:
@saedelman Do you agree with the proposal here or have something else/similar?

saedelman, 2018-02-08:
Yes, I agree with the proposal by @helmut64. I would name the functions consistent with the current approach, e.g. get_program_size() instead of minProgramSize(), but that's a minor detail. I don't understand the need for programAlignment(). I think if bytes need to be aligned than this alignment represents the minimum program size. Let me know what I am missing.

helmut64, 2018-02-08:
For minProgramSize() I don't know if this is the same as as programAlignment() which is usally 64-bit. I  have some experience with the STM MCUs Flash and more experience with the Winbond SPI Flash chips, however I don't know what it means for the other MCU vendors.

Anyways this is a minor detail, most important is that smaller writes are getting support.

davidsaada, 2018-08-06:
I'm trying to understand the issue here. FlashIAP has the `get_program_size` API, which maps to the minimal program unit (often called page size). If there are problems with specific boards, exposing a larger value for this unit, comparing to the board's capability, they should be handled on a per board basis (in flash_api level). Can you please explain how this differs from program alignment?

helmut64, 2018-08-06:
Hi David, I believe the alignment must be always on a multiple of the min. program size. So it is the same. However I don't have any overview of all implementations.
Regards Helmut

davidsaada, 2018-08-06:
> Hi  David, I believe the alignment must be always on a multiple of the min.  program size. So it is the same. However I don't have any overview of  all implementations. 

Not sure I follow. Can you specify an example with actual numbers for erase, program and alignment sizes from a board you know?
I for once, haven't seen this alignment unit size in any spec, but maybe that's a term known by another name.

helmut64, 2018-08-13:
I have some experience with WinBond Flash Chip, here are some important parameters:
- SektorSize (4 kByte, only a complete sector can be erased, or multiple depending on the chip)
  PS: I believe for the sector size a flash address needs to be provided because an MCU can have multiple flash areas with different flash parameters.
- MaxProgramSIze (256 bytes, if I need to write more I need to do multiple programs)
- MinProgramSIze (1 basically on any address,)
- WriteAlignment size (1 basically on any address)

On the STM32L4 MCUs series
- SektorSize (4kB or 8kB, depending on flash address/bank)
- MaxProgramSIze (256 bytes, on some other banks 512 bytes and on other banks 1024 bytes)
- MinProgramSIze (64-bit)
- WriteAlignment size (64-bit address)

Details see: stm32l4xx_hal_flash.h

Regards Helmut

davidsaada, 2018-08-13:
Thanks @helmut64. So, in both of your examples, the write alignment equals the min program size (which is essentially the program size). So the question is why you need the write alignment parameter?

helmut64, 2018-08-14:
@davidsaada yes I agree MinProgramSIze is identical to the WriteAlignment size. I believe we don't need the alignment parameter.

davidsaada, 2018-08-15:
> yes I agree MinProgramSIze is identical to the WriteAlignment size. I believe we don't need the alignment parameter.

Great. So - can we close this issue?

helmut64, 2018-08-16:
@davidsaada Hi David, I reviewed the FlashIAP API and implementation once more and I beliefe it is fine. This case can be closed.

Once more idea to discuss, many MCUs have multiple FLASH banks at different addresses with different features (sizes, block sizes, CRC), e.g. on the STM32L433
- 16 kB (SRAM2 with crc) @0x10000000
- 48 kB (SRAM1 with crc) @0x20000000

One option would be to add a bank parameter into the constructor with a default bank = 0, this allows to create the FlashIAP with the proper bank index and vendors can implement there additional banks if available.
Regards Helmut

closed 2018-10-02 by adbridge

helmut64, 2018-10-02:
FYI: finally I got some SAMD21 native flash programming experience, here are some data details:
- Programming size max 64 bytes, min 2 bytes
- Programming padding (even addresses)
- Sector size 256 bytes (Atmel calls it row size).