ARMmbed/mbed-os #5868: Unknown failure during mount - LittleFileSystem

moranpeker, 2018-01-17:
---------------------------------------------------------------
## Bug

**Target**
ALL

**Toolchain:**
GCC_ARM|ARM|IAR

**Actual behavior**
on LittleFileSystem there is no way to know if mount failed not : 

LittleFileSystem::LittleFileSystem(const char *name, BlockDevice *bd, ...............) {
    if (bd) {
        mount(bd);
    }
}

0xc0170, 2018-01-17:
@geky

geky, 2018-01-17:
If you pass the block device into the constructor, LittleFS will try to mount in the constructor, but it won't be able to return an error.

But, if you don't pass the block device into the constructor, you can pass it into the mount function, which does return errors if LittleFS fails to mount:
``` cpp
LittleFileSystem fs("fs");

int err = fs.mount(bd);
if (err) {
    printf("failed to mount: %d %s\n", err, strerror(-err));
}
```

One common pattern that may be helpful is to reformat if a mount fails:
https://github.com/ARMmbed/mbed-os-example-filesystem/blob/master/main.cpp#L78-L93

moranpeker, 2018-01-17:
i use storage selector so if i need partitions i have to pass the block device into the filesystem constructor (in filesystem_selector func) and i call mount function again because it will fail to try mount again.
other solution is to do unmount after filesystem constructor and then use mount function again to check if return an error. but of course its not the solution i am looking for.

geky, 2018-01-17:
Ah, in that case you can pass NULL as the block device to avoid mounting.

This isn't the prettiest and currently does feel like a workaround. I am planning on modifying the FileSystem classes to be initialized more like the BlockDevices (all params in the constructor, mount errors propagated to other functions), but since the FileSystem API is used heavily it may take a while to get changes in.

moranpeker, 2018-01-29:
i cant pass NULL if i need device partition.

geky, 2018-02-13:
Looking at storage selector, it looks like you can still pass NULL if you're using partitions.

Here's a change to the example in the storage selector readme:
``` cpp
// Get the block device
BlockDevice *bd = storage_selector();
bd->init();

// Create and initialize partitions
MBRBlockDevice::partition(bd , 1, 0x83, 0, 512*1024);
MBRBlockDevice::partition(bd , 2, 0x83, 512*1024, 1024*1024);
MBRBlockDevice part1(bd, 1);
MBRBlockDevice part2(bd, 2);

// Get unmounted filesystems
FileSystem *fs1 = filesystem_selector("fs1", NULL, 1);
FileSystem *fs2 = filesystem_selector("fs2", NULL, 2);

// Try to mount, or reformat on failure
int err = fs1.mount(&part1);
if (err) {
   err = fs1.reformat(&part1);
}

int err = fs2.mount(&part2);
if (err) {
   err = fs2.reformat(&part2);
}
```

moranpeker, 2018-02-15:
Hi @geky,
i am sure this will work . but its a workaround. 
Acually, it works fine even if i do not pass NULL parameter.
The problem is that there is a scenario you dont know if mount failed or not ,as i explain before :
LittleFileSystem::LittleFileSystem(const char *name, BlockDevice *bd, ...............) {
if (bd) {
mount(bd);
}
}

geky, 2018-02-15:
Ah are you referring to https://github.com/ARMmbed/mbed-os/issues/5871 or is there another issue with this approach?

moranpeker, 2018-02-15:
they are two different issued.
this issue is about that on LittleFileSystem constructor there is no check of mount return value. 
#5871 issue is about mount functionality .

geky, 2018-02-15:
Ah, so for this issues (no return value from mount). Does passing NULL work as a workaround?

If you pass NULL, constructor can not error, and can be taken from mount.

moranpeker, 2018-02-18:
Yes you right, passing null is a great workaround. **But the bug is still exist .** 
This is what we really want ?

moranpeker, 2018-07-03:
@geky fixed?

geky, 2018-07-03:
@moranpeker, sorry, not yet. Since there is a workaround it is low priority.

closed 2018-08-21 by moranpeker