ARMmbed/mbed-os #5869: How to export offline Project to KEIL uVision5 with custom scatterfile?

RaEyE, 2018-01-17:
### Description
- Type: Question
- Priority: Minor

---------------------------------------------------------------

## Question

Is there a possibility to use the offline export functionality of mbed-os to ignore scatterfiles of specific targets and automaticaly replace them with custom scatterfiles provided by the user?

## Description

Currently projects are exportet using the '.\mbed-os\tools\project.py' script.
example: `..\mbed-os\tools\project.py -m NUCLEO_F746ZG -i uvision5 --source . --source ..\mbed-os`

With the use of `.mbedignore` it is rather simple to ignore the addition of scatterfiles in general by ignoring any .sct file (e.g. `*.sct`), but it doesn't prevent the addition of
`--scatter ..\..\mbed-os\targets\TARGET_STM\TARGET_STM32F7\TARGET_STM32F746xG\device\TOOLCHAIN_ARM_STD\stm32f746xg.sct`
to the Linker configuration of KEIL uVision5, which effectively configures the IDE to use the scatterfile provided by mbed.

## Conclusion

If possible I'd like to know if the is a possibility to configure a custom scatterfile by the means of using `mbed_app.json` or the application of parameters when using `..\mbed-os\tools\project.py`.

Many Thank in advance
 - RaEyE

0xc0170, 2018-01-17:
@RaEyE  I assume  your use case is to follow mbed OS releases and just replace a scatter file? You can commit your scatter file to replace the one that is there by default (to update your version, rebase your changes).

@theotherjimmy

RaEyE, 2018-01-17:
@0xc0170 Thank you for your reply.
That would centainly be a valid and functional option to do so, but it would contradict with the current workflow.
Our projects based on mbed are oganized in such a way, the actual project is kept with all it's sources and accompanying files in a single folder parrallel to the actual mbed-os folder

| .
|-- project-dir/
|    |-- lib-a/
|    |-- lib-b/
|    |-- main.cpp
|    |-- .mbedignore
|    |__ mbed_app.json
|__ mbed-os/

All settings / configurations regarding the specific project are if anyhow possible organized with the use of mbed-os `*.json` and `.mbedignore` functionalities. The idea was to concentrate all the places where configuration changes regarding the project or general library settings are made in one place, namely the mbed export mechanic, but still keep library-dependant settings within the library's directory itself, therefore attached to the library.

TL/DR;
I was hoping for a solution to implement project-specific scatterfiles using the mbed export mechanic. If possible.

ohagendorf, 2018-01-18:
@RaEyE we are using the same princple. But up to now, the possibilities of the offline toolchain were enough for us.

In your situation a post processing after an export would be also a solution. The scatter file path is a single line in the Keil project config (XML) file. So, a XML search and replace could solve your problem.

theotherjimmy, 2018-06-18:
@RaEyE You could do as @0xc0170, check in your scatter file, and use `.mbedignore` in the project for the Mbed-OS provided scatter file.

theotherjimmy, 2018-06-20:
@RaEyE My mistake, Your project structure assumes that the project-directory would be adjacent to mbed-os. Why do you need that? You can accomplish the scatter file replacement much easier with mbed-os as a sub-directory in your project.

RaEyE, 2018-06-20:
@theotherjimmy Due to convenience. Just imagine having several different mbed-os based projects. Instead of having several projects "embedding" the exact same mbed-os, why not have a single mbed-os to which each project refers to.

Where necessary, these projects don't replace but overshadow parts or components of mbed-os by ignoring the original and then localy, in the project itself, replace the original by a modified version.
Where truly necessary mbed-os can be embedded as git submodule and project-specific changes can be implemented by a project-specific branch.

TL;DR - Many projects, but one mbed-os

theotherjimmy, 2018-06-20:
@RaEyE That's a pretty common theme. I'm working on improving that workflow. For many projects, one mbed-os I use symlinks. I recommend that as a workaround for the time being.

RaEyE, 2018-06-21:
@theotherjimmy As a workaround that would cenrtainley be ok, but I wouldn't recommend it as standard procedure.

For now I simply ignore the scatterfile and then manuly let Keil know which scatter to use.
Not a pistine solution, but for now it works. Since I use a batch to manually export new mbed-projects it shouldn't be that hard to add some lines in case to add / modify the corresponding Keil configurationline in the Keil Projectfile.