ARMmbed/mbed-os #5871: FAT FS partition magic field not set

moranpeker, 2018-01-17:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Related issue: `#abc`
- Priority: Major

---------------------------------------------------------------
## Bug

**Toolchain:**
GCC_ARM|ARM|IAR

**Actual behavior**
please follow this scenario while init file system :
init LittleFS .
init FatFS - mount failed -> reformat .
init back to LittleFS - mount failed. (expected to fail on muont and do reformat) 

**Steps to reproduce**
FS partition magic field not set .
while we init little file system the magic field is "LittleFS". when we init littleFS again after mount FATFS the mount succeed because the magic file is still "LittleFS".
we expect mount to failed.

geky, 2018-01-17:
In this case LittleFS just sees that the primary superblock is corrupt, assumes a write failed due to powerloss, and loads the backup superblock. Unfortunately this isn't great, since being formatted with the FAT filesystem may have corrupted the Little filesystem in a way that LittleFS can't detect.

If you erase both of the first two blocks, you should see the mount "fail" correctly.

At the moment I'm not sure what a better approach is though I'm open to suggestions.

We could scan for the FAT magic string first, but then would we need a set of all possible magic strings loaded into ROM just to mount LittleFS?

cmonr, 2018-01-17:
@geky Thoughts on adding an explicit "erase first two blocks" step as a part of the reformat, or would this be changing the behavior of the FatFS reformat step too much?

Alternatively, this could be documented with the expectation that when users are switching between the filesystems, the first two blocks must be erased.

moranpeker, 2018-01-29:
Hi,
mount function should be failed if we pass from Little to Fat or the opposite way . if the mount function failed we call reformat.

geky, 2018-02-04:
Currently you can guarantee this if you erase the block device when you reformat. For example:
``` cpp
int err = fs.mount(&bd);
if (err) {
    // no filesystem found, erase and reformat
    bd.init();
    bd.erase(0, bd.size());
    fs.reformat(&bd);
    bd.deinit();
}
```

Maybe we should erase the disk in our format function? We could have a parameter to disable a full erase.

One question is do Host OSs zero the disk during a format. Will need to check. Erasing the disk doesn't really help us if a user could end up in the same situation after formatting with a PC.

cmonr, 2018-02-04:
@geky I don't believe so. As an example, when you format a flash drive in WIndows, you can do a "quick" format, which afaik, only rebuilds the filesystem, erasing all file pointers, whereas a "non-quick" format presumably writes data (0s, 1s, noise, etc...) to the entire medium.

moranpeker, 2018-02-11:
the bug is that mount (under the specified scenario above) function is supposed to fail but it actually does not.

geky, 2018-02-11:
Mount is not fsck, it can't know if the disk has been messed with since boot. (But maybe we should provide a fsck function?).

We can tell if the disk has been erased, but before we add erase to the format functions we need to see what host OSs due, otherwise the problem still exists when formatted from a PC.

moranpeker, 2018-02-15:
Hi @geky  ,
i do not understand how mount function return success if it should fail because of switch between other file system types ( FAT and LITTLE).
as i can see it , mount function should cover this case too.

geky, 2018-02-15:
It's because FAT and LittleFS don't reside on the same locations in storage. The same issue happens on logging filesystems such as [SPIFFS](https://github.com/pellepl/spiffs), [JFFS2](https://sourceware.org/jffs2/jffs2-html/jffs2-html.html) and [YAFFS](https://yaffs.net).

It's only a side effect of the fact that FAT/NTFS/EXT/BTRFS all use the first block as a superblock that they notice when a different filesystem has been formatted on the block device.

Interestingly, `mkfs.fat` on Linux doesn't erase the whole block device, but it does erase the first 0x9800 bytes. I'm looking more into why this is.

geky, 2018-02-15:
One option is to encode a checksum into the sum of all revision counts (stored per directory) on the filesystem. This would allow you to catch if someone overwrote one of your directory blocks (and other time-travel bugs), but it would require summing all revision counts in the filesystem on mount and the extra code size to get this machinery working.

I'm not sure if it's worth that cost.

yossi2le, 2018-07-03:
@geky, looks to me like this issue can be closed now.

closed 2018-07-03 by 0xc0170