ARMmbed/mbed-os #5872: LWIP: alignment of Ethernet packet

hug-dev, 2018-01-17:
The `payload` pointer field of the `struct pbuf` structure is aligned when the structure is allocated.
However, when sending an Ethernet packet with the `ethernet_output` function, in file `mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/netif/lwip_ethernet.c`, it changes line 289 to make space for the Ethernet packet header and loses its alignment.
The `SIZEOF_ETH_HDR` define is set file `mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/lwip/prot/lwip_ethernet.h` line 84. Its value (14) makes it unaligned but 2 padding bytes can be added to make it aligned again. However, these padding bytes are removed before sending the packet to the Ethernet driver. This is shown in the Ethernet interface skeleton file `mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/include/lwip/prot/lwip_ethernet.h`, function `low_level_output` line 145.

Ultimately, this unaligned value led to a hard fault on the TARGET_CM3DS_MPS2, when compiling the mbed-os-example-client with the `release` profile of ARM, at a `LDM` instruction.

**My question is**: is this an issue in the LWIP stack or should the alignment be dealt with manually, i.e. allocating a new aligned structure and copying the packet there?

_P.S.:_ "alignment" here refers to a 32 bits boundary.

ashok-rao, 2018-01-19:
@SeppoTakalo .. any thoughts on this please?

SeppoTakalo, 2018-01-19:
CC: @kjbracey-arm @mikaleppanen Please check.

kjbracey-arm, 2018-01-19:
Where did the abort actually occur? What C code line generated the LDM instruction, and what was the target address?

None of the code should be relying on alignment - all relevant structures and pointers should be marked with a "packed" attribute, which would prevent the compiler generating LDM to an unaligned location, but maybe there's an omission.

hug-dev, 2018-01-19:
The error happened in the Ethernet driver of our target, CM3DS, at [line 621](https://github.com/ARMmbed/mbed-os/blob/c59c400093792f2bc73f3e7b6f78db37f88b08b0/targets/TARGET_ARM_SSG/TARGET_CM3DS_MPS2/SDK/smsc9220_eth.c#L621) of `targets/TARGET_ARM_SSG/TARGET_CM3DS_MPS2/SDK/smsc9220_eth.c` file.

The line that caused the error is:
`SMSC9220->TX_DATA_PORT = *pktptr;`
The following assembly is generated to get the `pktptr` pointed value:
`LDM r2!, {r3}` with `pktptr = r2`
and that causes an exception cause `pktptr = 0x20402CEE`

`pktptr` comes from the `data` argument of that function, called [here](https://github.com/ARMmbed/mbed-os/blob/c59c400093792f2bc73f3e7b6f78db37f88b08b0/features/FEATURE_LWIP/lwip-interface/lwip-eth/arch/TARGET_CM3DS_MPS2/mps2_emac.c#L214) and that last function is the one called by the `ethernet_output` function.

Should that `pktptr` pointer be marked as packed? We solved this problem locally by copying the content of `pktptr` in aligned words, byte per byte.

kjbracey-arm, 2018-01-19:
Yes, the lwIP configuration in mbed OS doesn't guarantee anything about the alignment of the data, so casting `q->payload` to an `unsigned int *` pointer like that is invalid. If you marked `pktptr` as packed that would fix it - but you'd need to locate an appropriate toolchain-independent macro to do that.

I can't immediately advise on the best packed macro, as my own code base tends to manually extract bytes without depending on packedness or endianness. You could also do the same.

gyuri-szing, 2018-01-19:
"the lwIP configuration in mbed OS doesn't guarantee anything about the alignment of the data"
Is this intentional or a configuration error?
As far as I can see by setting LWIP provides a method (e.g. setting macro values) to ensure proper packet buffer alignment. But then why are these not set properly?

kjbracey-arm, 2018-01-19:
Define "properly". lwIP would prefer to have its IP header 32-bit aligned (and it is in this case). This hardware would prefer (and actually expects) to have its Ethernet header 32-bit aligned. Some other hardware can be optionally configured to expect 2 bytes of leading padding.

The current set-up does permit a specific target driver to override lwIP config settings for its platform - the `lwipopts_conf.h` for this driver could probably adjust the config to get the alignment they want. (Although I'm not quite sure what that setting would be - I can see `ETH_PAD_SIZE`, but that's what they need).

That custom lwIP configuration is  viable if you have a single interface, but after #5558 that won't be possible - we're abstracting the system to have drivers not be directly linked to lwIP, and to support multi-interface use. 2 interfaces could disagree on the alignment they want.

gyuri-szing, 2018-01-19:
Thanks for the detailed explanation. We implemented the correct fix, which is the modification of the Ethernet driver.
From our side the issue can be closed.

hug-dev, 2018-01-19:
Thanks a lot for the explanation!

closed 2018-01-19 by hug-dev