ARMmbed/mbed-os #5879: About the target layer of nrf51822 and nrf52832

leibin2014, 2018-01-18:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Question
- Priority:  Minor

-----------------------------------------------------------------

## Question

Dear experts,
My company is using mbed for our IOT products. Now we want to replace the CPU from nrf51822 to nrf52832. 
I found the CPU architectures of both are very similar(Most of the registers are same). But in the mbed target layer different approaches are used. nrf51822 operates the CPU registers directly to implement spi/i2c/... api code. But nrf52832 uses SDK API. For example in the code of `targets/TARGET_NORDIC/TARGET_MCU_NRF51822/spi_api.c` it uses registers operation to implement functions. This is very simple and clear, and has no dependency with the SDK. But for nrf52832 it has strong dependency with SDK and interrupt is enabled.
Which approach in above both is better? I definitely prefer the registers directly operation approach as nrf51822 used. Why we don't implement the target api code(for example spi_api.c) of nrf52832 with the same approach as nrf51822? I have a plan to modify the target api code of nrf52832 the same as nrf51822 for our products. Is there any reason avoided me to do this?
Thanks!

screamerbg, 2018-01-18:
cc @ARMmbed/team-nordic  @marcuschangarm

nvlsianpu, 2018-01-18:
Hi - answer is simple - where we can we used SDK as simpler to use than pure register. In some cases it was impossible to use SDK driver. Some of driver were implemented before idea of using SDK. I is also easier to maintain the port - as the driver developer has much deeper knowledge about the each peripheral than I have. The last thin is that I have only a part of my time for mbed. If you see a gain in using registers instead of drivers - please explain why for each case. I can ques a few reasons too.
I'm open for proposition.

pan-, 2018-01-18:
@leibin2014 Initially Nordic HAL port was hand-rolled and only targeting the NRF51. During mbed OS 5 development, one of the new target being worked on was the NRF52. Both being supported by the Nordic SDK, it seems right to factor the HAL in common sources and rely on the Nordic SDK to handle abstraction over the raw peripherals/registers. Using the Nordic SDK means less maintenance for the Nordic team as well as benefiting from bug fixes from Nordic when the SDK is updated. 

>  I have a plan to modify the target api code of nrf52832 the same as nrf51822 for our products. Is there any reason avoid me to do this?

You may want to modify the hal implementation in two cases: 
* feature addition 
* bug fixes 

For both of this cases I advised you to either open an issue or a PR against this repository so you may benefits of your changes in subsequent mbed releases.

leibin2014, 2018-01-19:
@nvlsianpu@pan Thanks for your quick answers! The reasons for me planning to use pure registers are:
1. Nrf51822 use pure registers way and we got it working very well now. To make nrf52822 with the same way, it's easily to maintain both code.
2. With the SDK way the interrupt is required enabled for spi_api. But in my applications for some reason in some situation the interrupt is not allowed to be enabled. So in this kind of situation the nrf51822 spi api works fine, but nrf52832 doesn't.
3. Nrf52832 has strong dependency with SDK. This SDK version is just for softdevice s132 v2.0.0. If I want to update softdevice to some newer version. I have to update the SDK version also. But the new SDK version may be not suitable for the current mbed target api code. Then mbed spi/i2c... api code also need to be fixed and re-verified.
In summery, nrf52822 code has strong dependency on SDK. It's hard to control if we want to update the SDK or use some different way as SDK(for example disable interrupt). And the pure registers way of nrf51822 works very stable and anyway we already make it as this. And I noticed that nrf52832 had very similar registers sets with nrf51822, should not hard to port nrf51822 target api code to nrf52832. So I planned to make nrf52832 to use the same way to make above things simpler.
Yep, looks like it's not a simple thing as I expected.

leibin2014, 2018-01-22:
```This sounds like a bug; could you open an issue detailing the causes and consequences so we can provide a fix for everyone ?```
@pan- I'm not sure whether this is a bug. Maybe just different approaches to implement spi api. We can use interrupt way or just don't use interrupt. I have already changed my applications to fit the interrupt way now. But interrupt enabling is still a must for current spi api implementation of nrf52.
```Note that there is an ongoing effort to update Nordic HAL and softdevice.```
This is good news. It means we can use new softdevice later. We can also see as expected once we want to update softdevice we have to update SDK. And since the mbed target code has strong dependency with SDK, a lot of mbed target code is also required to be fixed.
I created this issue inorder to get more information/history about the different approaches for nrf51 and nrf52. I have already achieved the information. I will close this issue later if there is no further considering.
Thanks for all the information you both provided!

closed 2018-01-24 by leibin2014