ARMmbed/mbed-os #5881: Can not set a Serial charater format of 7 data bit + parity on STM32 board after patch #4189.

Roietronics, 2018-01-18:
### Description
Can not set a Serial character format of 7 data bit + parity on STM32 board after patch #4189, STM32s 
Serial does not properly handle parity bits.
Please see Discussion post  on mbed bug forum with the same title for more details
- Type: Bug 
- Related issue: 
- Priority:| Major 
Pull request with patch coming

---------------------------------------------------------------
## Bug

**Target**
SMT32?????

**Toolchain:**
all
**Toolchain version:**
all
**mbed-cli version:** 
> 5.4.7
**mbed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**
8 bit serial character with 7 data bits and a Parity bit as the msb
**Actual behavior**
9 bit serial character with 7 data bit a parity bit as the msb and a random bit between the 7th data bit and the parity 
Received characters are 9 bits with a random msb
**Steps to reproduce**
Call the Serial format method with the follow argument (7,SerialBase::Even, 1) for a target board using a microcontroller from the STM32 family
Send a character and observe the bit pattern on the line
Send a character to a device which expects characters with 7 data bit and a parity bit and observe framing error
Send a character which require the parity bit to set for even parity to the same device and observe a parity error 

read a character sent to the board from device which transmits a 8 bit character with 7 data bits and parity and observed 9 bit character high since its was the stop bit

Send a string of character to the board from a device sending 8 bit characters with one stop bit and observe a line hang since there is no stop bit before the next start bit. The stop bit having been clocked in as UART as the msb of a nine bit character
----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

0xc0170, 2018-01-18:
Is this this commit that is causing a problem: https://github.com/fmanno/mbed-os/commit/5e98da14d328e5f466fc41ce7f894e0bcce13e96 ?

@ARMmbed/team-st-mcd

bcostm, 2018-01-18:
Which board are you using. Because the 7bit mode is not supported on all STM32 devices.

Roietronics, 2018-01-18:
The point of this entry is to point out that a serial data stream using character with 7 data bits and parity is uses eight bit not seven bit characters and support for a 7 bit mode on the device is not an issue.
Any device with an eight bit mode should be able to support a 7 data bit + parity serial data stream.
For mbed-os releases prior to the patch I was able to establish 7 data bit + parity data stream using  both a STM32F411RE on a mDot board and a STM32F468RE on a Nucelo board which communicated with a SDI-12 sensor. Neither of these devices support a 7 bit mode. For mbed-os release after the patch I was not able to format a data stream which the SDI-12 device would accept. Using an Oscilloscope on the serial line verified that 8 bit characters were being sent prior to the patch and after the patch only 9 bit character could be transmitted and there was no set of arguments to the format method which would allow the transmission and reception of 8 bit characters if a parity option other than none was specified.

Roietronics, 2018-01-18:
0xc0170 yes that is the commit which is causing the problem

0xc0170, 2018-01-19:
@Roietronics Why did you close?

bcostm, 2018-01-19:
What I see in today's code is that the 8bit data format is selected if the 7bit is not supported. This is the case for the STM32F411 which does not support the 7bit mode. I think it would be more appropriate to exit with an error instead. What do you think @LMESTM ?

Today there is:
```
    switch (data_bits) {
        case 9:
            MBED_ASSERT(parity == UART_PARITY_NONE);
            obj_s->databits = UART_WORDLENGTH_9B;
            break;
        default:
        case 8:
            if (parity != UART_PARITY_NONE) {
                obj_s->databits = UART_WORDLENGTH_9B;
            } else {
                obj_s->databits = UART_WORDLENGTH_8B;
            }
            break;
#if defined UART_WORDLENGTH_7B
        case 7:
            if (parity != UART_PARITY_NONE) {
                obj_s->databits = UART_WORDLENGTH_8B;
            } else {
                obj_s->databits = UART_WORDLENGTH_7B;
            }
            break;
#endif
    }
```

I propose:
```
    switch (data_bits) {
        case 9:
            MBED_ASSERT(parity == UART_PARITY_NONE);
            obj_s->databits = UART_WORDLENGTH_9B;
            break;
        default:
        case 8:
            if (parity != UART_PARITY_NONE) {
                obj_s->databits = UART_WORDLENGTH_9B;
            } else {
                obj_s->databits = UART_WORDLENGTH_8B;
            }
            break;
        case 7:
#if defined UART_WORDLENGTH_7B
            if (parity != UART_PARITY_NONE) {
                obj_s->databits = UART_WORDLENGTH_8B;
            } else {
                obj_s->databits = UART_WORDLENGTH_7B;
            }
#else
            error("7bit data format not supported on this platform");
#endif
            break;
    }
```

Roietronics, 2018-01-19:
Was not supposed to be closed. I am not sure what cause the close ?????

Roietronics, 2018-01-19:
The problem with bcostm's purposed is that the if block in the if else statement enclosed with in the #if defined block is a valid format on platforms which  do not support 7 bit character since the word length is 8 bits not 7 bits for this case.
This is the format needed to support protocol such as the SDI-12 data logger protocol which use serial data stream with 7 data bit + a parity bit (i.e.  8 bit) characters
Prior to the patch in question this format could be created by specifying (8, SerialBase::Even, 1) as the arguments for the format command.
After the patch these arguments result in a 9 bit character instead of an 8 bit character.  After the patch there is no set of arguments to the format function which will create 8 bit characters for a 7 data bits + 1 parity bit serial data stream. 
The patch has effectively changed the meaning of the format method's first argument from character length in bits to number of data bits in the character.  The two values are equal for parity equal to none but differ by one bit for all other parity options

bcostm, 2018-01-22:
OK.

Can you try this :

```
    switch (data_bits) {
        case 9:
            MBED_ASSERT(parity == UART_PARITY_NONE);
            obj_s->databits = UART_WORDLENGTH_9B;
            break;
        default:
        case 8:
            if (parity != UART_PARITY_NONE) {
                obj_s->databits = UART_WORDLENGTH_9B;
            } else {
                obj_s->databits = UART_WORDLENGTH_8B;
            }
            break;
        case 7:
            if (parity != UART_PARITY_NONE) {
                obj_s->databits = UART_WORDLENGTH_8B;
            } else {
#if defined UART_WORDLENGTH_7B
                obj_s->databits = UART_WORDLENGTH_7B;
#else
                error("7bit data format without parity is not supported on this platform");
#endif
            }
            break;
    }
```

> This is the format needed to support protocol such as the SDI-12 data logger protocol which use serial data stream with 7 data bit + a parity bit (i.e. 8 bit) characters

And use the format `(7, SerialBase::Even, 1) `

cmonr, 2018-01-23:
PR merged.

closed 2018-01-23 by cmonr