ARMmbed/mbed-os #5883: Nordic RF52 does not correctly support dynamically created SPI/I2C (TWI) devices

NeilMacMullen, 2018-01-18:
Mbed 5.6/5.7
Target: Nordic RF52832

The 52832 supports up to 3 simultaneous SPI/I2C buses.   Our board requires 1xI2C and 3xSPI so we statically allocate an I2C and SPI bus then dynamically allocate the remaining bus controller as required. 

Unfortunately the reallocation of both I2C and SPI TWO blocks is broken.  This code will quickly fault...

     I2C *i2c = new I2C(p16, p15);
     SPI *storageSPI = new SPI(p9, p10, p14);
     while (true)
     {
    SPI *gpsSpi = new SPI(p29, p28, p30, p31);
     delete gpsSpi;
    SPI *espSpi = new SPI(p6, p8, p7);
    delete espSpi;
    }

There is a comment in   spi_init that indicates that the mbed  SPI class does not call spi_free from the destructor. This is true but even if the call is made and the 'bodge' code in spi_init is removed, it doesn't seem possible to correctly reuse previously-freed buses.

NeilMacMullen, 2018-01-18:
Adding @MarceloSalazar Might be fixed in https://github.com/ARMmbed/mbed-os/pull/5726/files/b1fa9e9820dd4b71ec4e4793c2e20373fc47e067..d2e20e673da9f06c8f948e81a0d08d93247c2c1d

MarceloSalazar, 2018-01-18:
Related to https://github.com/ARMmbed/mbed-os/issues/5883
cc @dlfryar @marcuschangarm

NeilMacMullen, 2018-01-18:
Further info... If spi_free(&_spi); is called in the SPI destructor and the first block of code in spi_init is removed then it is possible to reallocate a previously deleted SPI block but it can't be read/written.  The spi_free code appears to be sufficient to clear any hardware state in the TWI block and it then marks the initialised flag as false which means the hardware should be set up correctly on reallocation but perhaps there is a missing step?  Needs input from Nordic.

cmonr, 2018-01-18:
@NeilMacMullen I wouldn't bet on #5726. Those changes are slated to enter a feature branch, not the main branch, and there's no eta as to when they'll be merged together.

cmonr, 2018-01-18:
@ARMmbed/team-nordic

cmonr, 2018-01-19:
Reproduced on a NRF52-DK.

screamerbg, 2018-01-30:
cc @nvlsianpu

marcuschangarm, 2018-03-08:
We have a PR up for our new SPI and I2C driver on our feature branch: https://github.com/ARMmbed/mbed-os/pull/6296

You can follow SDK 14.2 updates here: https://os.mbed.com/forum/upcoming-features/topic/29477/

0xc0170, 2018-05-18:
The feature branch shared above was integrated, it's now on master. This should be resolved

closed 2018-05-18 by 0xc0170