ARMmbed/mbed-os #5885: Renesas GR-PEACH (RZ_A1H) is missing TRNG, Entropy

JanneKiiskila, 2018-01-19:
### Description
- Type: Bug or Enhancement | Question
- Related issue: 
- Priority:  Major 

---------------------------------------------------------------
## Bug

**Target**
 RZ_A1H

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**
N/A

**mbed-cli version:** 
(`mbed --version`)
N/A

**mbed-os sha:**
(`git log -n1 --oneline`)
C:\Users\jankii01\mbed\mbed-os-example-client\mbed-os>git log -n1 --oneline
c59c40009 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #5832 from geky/fix-truncate-zero

**DAPLink version:**
N/A

**Expected behavior**

mbed-os-example-client compiles

**Actual behavior**

You must add the flags to disable TRNG/Entropy checks to `mbed_app.json`.

**Steps to reproduce**
```
git clone git@github.com:ARMmbed/mbed-os-example-client.git
- get Mbed Connect security.h -file
mbed compile -m RZ_A1H  -t <your toolchain>
```

works with: 

```
    "macros": ["MBEDTLS_USER_CONFIG_FILE=\"mbedtls_mbed_client_config.h\"" ,"MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES", "MBEDTLS_TEST_NULL_ENTROPY"],

```
in `mbed_app.json`, but that's not secure.

This will be needed for Mbed Cloud Client -enablement.

0xc0170, 2018-01-19:
I've noticed their new target contains trng via i2c sensor.

@ARMmbed/team-renesas Please review

TomoYamanaka, 2018-01-19:
GR-PEACH doesn't support TRNG. GR-LYCHEE that is renesas new target supports TRNG(via i2c).

TomoYamanaka, 2018-01-20:
Could you tell me where the `mbed_app.json` file that you mentioned?

TomoYamanaka, 2018-01-25:
@JanneKiiskila 
Is the file that you pointed out the below file?
https://github.com/ARMmbed/mbed-os-example-client/blob/master/mbed_app.json

JanneKiiskila, 2018-01-25:
Yes, that's the file one needs to modify. There are also more similar files under `/configs` folder that can also be used.

TomoYamanaka, 2018-01-29:
@JanneKiiskila 
I submitted PR for fix this issue. Could you please review it.

JanneKiiskila, 2018-01-29:
Hi @TomoYamanaka ,

I'm afraid that one produces a gazillion warnings when being used and by default we should only support boards that support the TRNG (it is a HW security requirement from the Client point of view).

There is another solution we are working on as well, we'll have to see when we can get that done - I think that one would resolve the issue for boards w/o TRNG.

TomoYamanaka, 2018-01-29:
@JanneKiiskila 

Thank you for your comments.
Should I remove the above my PR and wait another solution that you mentioned?

JanneKiiskila, 2018-02-13:
Hi,

yes, I think that's the best. We'll implement the support for non-TRNG boards via a feature called SOTP.

closed 2018-02-13 by JanneKiiskila