ARMmbed/mbed-os #5886: Renesas RZ_A1H missing NVIC_SystemReset

JanneKiiskila, 2018-01-19:
### Description
- Type: Bug or Enhancement | Question
- Related issue: 
- Priority:  Major 

---------------------------------------------------------------
## Bug

**Target**
 RZ_A1H

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**
N/A

**mbed-cli version:** 
(`mbed --version`)
N/A

**mbed-os sha:**
(`git log -n1 --oneline`)
C:\Users\jankii01\mbed\mbed-os-example-client\mbed-os>git log -n1 --oneline
c59c40009 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #5832 from geky/fix-truncate-zero

**DAPLink version:**
N/A

**Expected behavior**

mbed-os-example-client compileshttps://connector.mbed.com/

**Actual behavior**

Doesn't compile, due to missing  reboot definition.

```
  ^~~~~~~
.\pal\Source\Port\Reference-Impl\mbedOS\RTOS\pal_plat_rtos.cpp: In function 'void pal_plat_osReboot()':
.\pal\Source\Port\Reference-Impl\mbedOS\RTOS\pal_plat_rtos.cpp:243:22: error: 'NVIC_SystemReset' was not declared in this scope
     NVIC_SystemReset();
```

**Steps to reproduce**
```
git clone git@github.com:ARMmbed/mbed-os-example-client.git
- get Mbed Connect security.h -file (https://connector.mbed.com/)
mbed compile -m RZ_A1H  -t <your toolchain>
```

This will be needed for Mbed Cloud Client -enablement.

0xc0170, 2018-01-19:
`pal_plat_rtos.cpp` - I would like to see what header files they include. Is this only to renesas platform, the rest are compiling?

cmsis_nvic.c defines those functions, it is defined for renesas. Its declared in `cmsis_nvic.h`. This header file might be missing to be included to get to the application space? in `cmsis.h` there should be `#include "cmsis_nvic.h"`.

@ARMmbed/team-renesas please review

0xc0170, 2018-01-19:
fix #5890 referenced above, please review

TomoYamanaka, 2018-01-19:
@0xc0170 

Thank you. I think this issue will be fixed by PR #5890.

PR #5890 (2018-01-23): RZ_A1H: cmsis nvic include fix
To get cmsis nvic definitions, this header file should be included
in cmsis.h file.

Fixes #5886

@JanneKiiskila @ARMmbed/team-renesas
commit 5554ab28 (2018-01-23): RZ_A1H: cmsis nvic include fix

To get cmsis nvic definitions, this header file should be included
in cmsis.h file.

Fixes #5886
closed 2018-01-23 by cmonr