ARMmbed/mbed-os #5887: ff_lpc546xx:  Library name 'platform' is not unique

gorazdko, 2018-01-19:
Compiling example https://os.mbed.com/teams/mbed/code/mbed_blinky/ in online compiler for target FF_LPC546XX results in error:

Error: Library name 'platform' is not unique (defined in '/extras/mbed_7130f322cb7e/TARGET_LPC546XX/mbed_lib.json' and '/extras/mbed_7130f322cb7e/TARGET_FF_LPC546XX/mbed_lib.json')

Compiling examples with mbed-os  works fine.

See PR #5876 for details

0xc0170, 2018-01-19:
IS this happening only in the online compiler ? Offline all works?

gorazdko, 2018-01-24:
mbed_lib.json is:

```
{
    "name": "platform",
    "config": {
        "stdio-convert-newlines": {
            "help": "Enable conversion to standard newlines on stdin/stdout",
            "value": false
        },

        "stdio-baud-rate": {
            "help": "Baud rate for stdio",
            "value": 9600
        },

        "stdio-flush-at-exit": {
            "help": "Enable or disable the flush of standard I/O's at exit.",
            "value": true
        },

        "default-serial-baud-rate": {
            "help": "Default baud rate for a Serial or RawSerial instance (if not specified in the constructor)",
            "value": 9600
        }
    },
    "target_overrides": {
        "EFM32": {
            "stdio-baud-rate": 115200
        },
        "EFR32": {
            "stdio-baud-rate": 115200
        }
    }
}
```

I cannot compile with mbed-cli because the target name is not found which is another bug:
```

C:\Users\Gorazd\Desktop\mbed-os-2-fflpc546xx\mbed-error_reporduce_in_cli\mbed_blinky\mbed>mbed update 7130f322cb7e
[mbed] Updating library "mbed" to rev #7130f322cb7e

C:\Users\Gorazd\Desktop\mbed-os-2-fflpc546xx\mbed-error_reporduce_in_cli\mbed_blinky\mbed>cd ..

C:\Users\Gorazd\Desktop\mbed-os-2-fflpc546xx\mbed-error_reporduce_in_cli\mbed_blinky>mbed compile -t ARM -m FF_LPC546XX
WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
usage: make.py [-h] [-m MCU] [-t TOOLCHAIN] [--color] [--cflags CFLAGS]
               [--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
               [--profile PROFILE] [--app-config APP_CONFIG] [-p PROGRAM]
               [-n PROGRAM] [-j JOBS] [-v] [--silent] [-D MACROS]
               [-S [{matrix,toolchains,targets}]] [-f GENERAL_FILTER_REGEX]
               [--stats-depth STATS_DEPTH] [--automated] [--host HOST_TEST]
               [--extra EXTRA] [--peripherals PERIPHERALS]
               [--dep DEPENDENCIES] [--source SOURCE_DIR]
               [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
               [-d DISK] [-s SERIAL] [-b BAUD] [-L] [--rpc] [--usb] [--dsp]
               [--testlib] [--build-data BUILD_DATA] [-l LINKER_SCRIPT]
make.py: error: argument -m/--mcu: FF_LPC546XX is not a supported MCU. Supported MCUs are:
ARCH_BLE,              ARCH_BLE_BOOT,         ARCH_BLE_OTA,
ARCH_GPRS,             ARCH_LINK,             ARCH_LINK_BOOT,
ARCH_LINK_OTA,         ARCH_MAX,              ARCH_PRO,
ARM_BEETLE_SOC,        ARM_CM3DS_MPS2,        ARM_IOTSS_BEID,
ARM_MPS2_M0,           ARM_MPS2_M0P,          ARM_MPS2_M1,
ARM_MPS2_M3,           ARM_MPS2_M4,           ARM_MPS2_M7,
B96B_F446VE,           BLUEPILL_F103C8,       DELTA_DFBM_NQ620,
DELTA_DFCM_NNN40,      DELTA_DFCM_NNN40_BOOT, DELTA_DFCM_NNN40_OTA,
DELTA_DFCM_NNN50,      DELTA_DFCM_NNN50_BOOT, DELTA_DFCM_NNN50_OTA,
DISCO_F051R8,          DISCO_F100RB,          DISCO_F303VC,
DISCO_F334C8,          DISCO_F401VC,          DISCO_F407VG,
[... 45 lines cut ...]
NUMAKER_PFM_M487,      NUMAKER_PFM_NANO130,   NUMAKER_PFM_NUC472,
NZ32_SC151,            OC_MBUINO,             RBLAB_BLENANO,
RBLAB_BLENANO_BOOT,    RBLAB_BLENANO_OTA,     RBLAB_NRF51822,
RBLAB_NRF51822_BOOT,   RBLAB_NRF51822_OTA,    REALTEK_RTL8195AM,
RO359B,                RZ_A1H,                SAMD21G18A,
SAMD21J18A,            SAMG55J19,             SAML21J18A,
SAMR21G18A,            SARA_NBIOT_EVK,        SEEED_TINY_BLE,
SEEED_TINY_BLE_BOOT,   SEEED_TINY_BLE_OTA,    SILICA_SENSOR_NODE,
SSCI824,               Super_Target,          TB_SENSE_1,
TB_SENSE_12,           TEENSY3_1,             TMPM066,
TY51822R3,             TY51822R3_BOOT,        TY51822R3_OTA,
UBLOX_C027,            UBLOX_C030_N211,       UBLOX_C030_U201,
UBLOX_EVA_NINA,        UBLOX_EVK_NINA_B1,     UBLOX_EVK_ODIN_W2,
UBRIDGE,               USENSE,                VBLUNO51,
VBLUNO51_BOOT,         VBLUNO51_LEGACY,       VBLUNO51_OTA,
VBLUNO52,              VK_RZ_A1H,             WALLBOT_BLE,
WALLBOT_BLE_BOOT,      WALLBOT_BLE_OTA,       WIZWIKI_W7500,
WIZWIKI_W7500ECO,      WIZWIKI_W7500P,        XADOW_M0,
XBED_LPC1768,          XDOT_L151CC
[mbed] ERROR: "c:\python27\python.exe" returned error code 2.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Users\Gorazd\Desktop\mbed-os-2-fflpc546xx\mbed-error_reporduce_in_cli\mbed_blinky\.temp\tools\make.py -t ARM -m FF_LPC546XX --source . --build .\BUILD\FF_LPC546XX\ARM" in "C:\Users\Gorazd\Desktop\mbed-os-2-fflpc546xx\mbed-error_reporduce_in_cli\mbed_blinky"
---

C:\Users\Gorazd\Desktop\mbed-os-2-fflpc546xx\mbed-error_reporduce_in_cli\mbed_blinky>
```

0xc0170, 2018-01-24:
From the log, I cannot spot `FF_LPC546XX` there.
But looking at the sha, latest mbed 2 version - it contains this target - https://os.mbed.com/users/mbed_official/code/mbed/file/7130f322cb7e/targets.json/ 😩 

@theotherjimmy can you help?

theotherjimmy, 2018-01-24:
@gorazdko This looks like the release contains a complete mbed implementation for both LPC546xx and FF_LPC546xx. That is a bit strange, because FF_LPC546xx does not contain the key "release_version". @adbridge This seems like a release mistake, could we look into correcting it?

@0xc0170 @gorazdko I need to update the mbed 2 tools to parse the new target. I'll do that this week. After I do that, you should be able to reproduce this offline as well.

0xc0170, 2018-01-25:
> @gorazdko This looks like the release contains a complete mbed implementation for both LPC546xx and FF_LPC546xx. That is a bit strange, because FF_LPC546xx does not contain the key "release_version". @adbridge This seems like a release mistake, could we look into correcting it?

Good catch 👍

@gorazdko https://github.com/ARMmbed/mbed-os/pull/5344 - did not have release version labeled, intentional or should be fixed?

theotherjimmy, 2018-01-25:
@0xc0170 Adding a release version would be a mistake as it would enforce this compiler error.

gorazdko, 2018-02-02:
Hi, what is the status of this ticket?

screamerbg, 2018-02-02:
@gorazdko @theotherjimmy The target definition for FF_LPC546XX in target.json must not inherit from LPC546XX, otherwise for Mbed 2.0/Classic builds, it won't compile due to duplicate symbols errors, e.g. due to the inheritance from existing target, both target build directories in the Mbed 2.0 release TARGET_FF_LPC546XX and TARGET_LPC546XX, causing duplicate linker errors. I think that this is why you're seeing the mbed_lib.json errors as well, because the build system tries to include both binary build directories due to the target inheritance.

theotherjimmy, 2018-02-02:
@screamerbg I looked and the FF_... does not have release version 2 attached to it. My understanding was that: therefore it should not get it's own build of mbed 2

screamerbg, 2018-02-02:
The parent target has 2 and 5 release versions, therefore it gets both. https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json

Also, we want Mbed 2 releases for this board :)

gorazdko, 2018-02-02:
> The target definition for FF_LPC546XX in target.json must not inherit from LPC546XX

So #5876 is OK then?

screamerbg, 2018-02-02:
Yes. The PR is correct. You would have to wait for the next Mbed 2.0/Classic release until it starts working in the online IDE.

adbridge, 2018-02-06:
@gorazdko Just FYI the next Mbed 2.0 release is in 2 weeks time :)

ashok-rao, 2018-02-27:
@theotherjimmy ..Regarding: https://github.com/ARMmbed/mbed-os/issues/5887#issuecomment-360217604 .. looks like the issue is still reproducible in the online IDE.. have the new mbed 2.0 tools been released with this target added? Thanks.

theotherjimmy, 2018-02-27:
@ashok-rao While the online compiler and the mbed 2 tools are the same, I don't have to release the mbed 2 tools OR update the online compiler for that PR to take affect.

theotherjimmy, 2018-02-27:
I'm having trouble reproducing the error. What's the name of the FF_LPC546xx in the online compiler?

ashok-rao, 2018-02-27:
L-Tek FF-LPC546XX . I'm on v161 of Mbed lib and just compiling blinky on Classic (not OS 5).

[image]

gorazdko, 2018-02-27:
https://os.mbed.com/teams/mbed/code/mbed_blinky/

theotherjimmy, 2018-02-27:
Weird, I don't see the L-Tek FF-LPC546XX.

gorazdko, 2018-02-27:
because the platform is not opened yet
https://os.mbed.com/platforms/L-TEK-FF-LPC546XX/

theotherjimmy, 2018-02-27:
Thanks. Now I can reproduce it.

theotherjimmy, 2018-02-27:
Hm, No I can't. I get a "The Requested Build is not available" error.

ashok-rao, 2018-02-27:
v160 of Classic (not 161). I edited the comment earlier..

gorazdko, 2018-02-27:
Note:  NXP LPCxpresso54608  (LPC546XX) builds fine. Thats the platform it is clashing with.

theotherjimmy, 2018-02-27:
@ashok-rao version 160 contains the bug.

ashok-rao, 2018-02-27:
Yep @theotherjimmy, and v161 doesn't seem to build.. @adbridge .. we are facing issues with v161 of the Classic build.. can you please have a look?

adbridge, 2018-02-28:
@ashok-rao we don't have a v161 of mbed 2 , are you referring to the version in the repo rather than the version in mbed-os.h ?

adbridge, 2018-02-28:
So I tried it and get:
Error: Library name 'platform' is not unique (defined in 
/extras/mbed_5571c4ff569f/TARGET_LPC546XX/mbed_lib.json' 
and '/extras/mbed_5571c4ff569f/TARGET_FF_LPC546XX/mbed_lib.json'
which is the same as reported above. 

mbed2 compiles work differently to mbed-os as I believe all the files get copied into one directory, which means any clashes will show up immediately. 
I don't know if this means the target inheritance on FF_LPC546XX is still wrong somehow or there is a tool issue. I don't currently know enough about how the target stuff works to help. 
@0xc0170  do you have any ideas on this one?

theotherjimmy, 2018-02-28:
@adbridge Yes, we are referring to the commit in the repo. We can't open the `mbed.h` to view the version.

theotherjimmy, 2018-02-28:
> I don't know if this means the target inheritance on FF_LPC546XX is still wrong somehow or there is a tool issue.

Actually, it means we have not made a release of Mbed 2 that included #5876

ashok-rao, 2018-03-01:
Thanks both.. @adbridge .. when do you think v161 of Mbed 2 will be released with #5876 included?

adbridge, 2018-03-01:
@ashok-rao The next release of mbed 2 is due alongside 5.7.7 in 2 weeks time

ashok-rao, 2018-03-27:
@theotherjimmy @adbridge .. looks like the issue is still reproducible..
```
Error: Library name 'platform' is not unique (defined in '/extras/mbed_994bdf8177cb/TARGET_LPC546XX/mbed_lib.json' and '/extras/mbed_994bdf8177cb/TARGET_FF_LPC546XX/mbed_lib.json')
```
I tried all these revisions below:
[image]

@gorazdko .. I'm running out of ideas for what else we can try to fix this.

HarishkumarBringiramakannu, 2018-06-19:
Hi @ashok-rao , I am using STM32l072 microcontroller. Now I am trying to compile DS18B20 Temperature sensor program. But I am getting the same error what you got. My error is,
"Error: Library name 'platform' is not unique (defined in '/tmp/chroots/ch-2619ff6b-a6f9-4624-81ff-1ac9c3c14ea0/extras/mbed/TARGET_DISCO_L072CZ_LRWAN1/mbed_lib.json' and '/tmp/chroots/ch-2619ff6b-a6f9-4624-81ff-1ac9c3c14ea0/src/mbed-os/platform/mbed_lib.json')"

Did you clear this ? Could you say how did you clear this ?

gorazdko, 2018-06-22:
Hi, what is the status of this Issue? Is there anything we could do to resolve this Issue?

@adbridge 
@screamerbg 
@theotherjimmy

HarishkumarBringiramakannu, 2018-06-22:
@adbridge @gorazdko Thanks for your replies. So there is no other way to rectify this ?I am using online compiler.

theotherjimmy, 2018-06-22:
There's no way that I know of to work around this in the online compiler. My suggestion was going to be "go offline and delete some stuff from the mbed 2 directory". Sorry.

adbridge, 2018-07-03:
Well there is a duplication of mbed_lib.json files so how does that even happen? Surely there should only be one of these per application ??

theotherjimmy, 2018-07-03:
I think the release process creates many duplicates...

screamerbg, 2018-09-19:
@gorazdko @adbridge Is this still an issue? At least from my perspective, this is no longer an issue about the duplicate platform definition, but rather release process issues.

theotherjimmy, 2018-10-01:
Just checked with the most recent mbed 2 release, it's still an issue.

theotherjimmy, 2018-10-01:
https://github.com/ARMmbed/mbed-os/pull/8291 should finish this off.

ruevs, 2018-11-27:
Sorry for barging in on this but there is at least one more target that has the same problem in the online compiler - Xadow-M0.
Here is a blinky reproducing the problem with MBED 2 rev 177:3a7713b1ebdc  
https://os.mbed.com/users/ruevs/code/xadow_m0_blinky/

The error message is:

> Error: Library name 'platform' is not unique (defined in '/tmp/chroots/ch-32cf0677-917d-4063-970e-6b43474ba95e/extras/mbed/TARGET_LPC11U35_501/TOOLCHAIN_ARM_MICRO/mbed_lib.json' and '/tmp/chroots/ch-32cf0677-917d-4063-970e-6b43474ba95e/extras/mbed/TARGET_XADOW_M0/TOOLCHAIN_ARM_MICRO/mbed_lib.json')

This problem was introduced in:
>Rev:  142:4eea097 10 May 2017
>
> Release 142 of the mbed library
> 
> Ports for Upcoming Targets
> 
> Fixes and Changes
> 
> 4059: [Silicon Labs] Rename targets https://github.com/ARMmbed/mbed-os/pull/4059
> 4187: [NCS36510] Reduce default heap size allocated by IAR to 1/4 of RAM https://github.com/ARMmbed/mbed-os/pull/4187
> 4225: fixed missing device_name for xDot and removed progen https://github.com/ARMmbed/mbed-os/pull/4225
> 4251: Fix C++11 build error w/ u-blox EVK-ODIN-W2 https://github.com/ARMmbed/mbed-os/pull/4251
> 4236: STM32 Fixed warning related to __packed redefinition https://github.com/ARMmbed/mbed-os/pull/4236
> 4190: LPC4088: Enable LWIP feature https://github.com/ARMmbed/mbed-os/pull/4190
> 4260: Inherit Xadow M0 target from LPC11U35_501 https://github.com/ARMmbed/mbed-os/pull/4260
> 4249: Add consistent button names across targets https://github.com/ARMmbed/mbed-os/pull/4249

theotherjimmy, 2018-11-27:
Thanks @ruevs It's probably #4260 that did it, if I were to guess.

ruevs, 2018-11-28:
I suspected #4260 as well but I am not at all familiar with the build system. The text I quoted above is just the commit message for rev. 142, there could have been other changes
I first noticed this almost a year ago (https://os.mbed.com/questions/79756/Xadow-M0-does-not-compile-with-mbed-abov/) but since I was not actively using the board I did not pursue it seriously.

robin604, 2019-10-02:
issue solved

gorazdko, 2019-10-06:
>  issue solved??

yes, issue has been solved

closed 2019-10-06 by gorazdko

ruevs, 2019-10-08:
No it has not been fixed.

Building the TARGET_XADOW_M0 fails on MBED 2 rev 172 (https://os.mbed.com/users/mbed_official/code/mbed/builds/65be27845400) with:

 Error: Library name 'platform' is not unique (defined in '/tmp/chroots/ch-61c13650-bafb-40ee-bc8b-5842d8ff86d8/extras/mbed/TARGET_LPC11U35_501/TOOLCHAIN_ARM_MICRO/mbed_lib.json' and '/tmp/chroots/ch-61c13650-bafb-40ee-bc8b-5842d8ff86d8/extras/mbed/TARGET_XADOW_M0/TOOLCHAIN_ARM_MICRO/mbed_lib.json')

https://os.mbed.com/users/ruevs/code/xadow_m0_blinky/

gorazdko, 2019-10-08:
> No it has not been fixed.

sorry, i was refering to ff-lpc546xx. Meybe open a new issue for your target...

ruevs, 2019-10-08:
Done