ARMmbed/mbed-os #5888: Timer return function casting issue

akrnvm, 2018-01-19:
### Description
- Type: Bug | Question
- Related issue: `#timer`
- Priority: Blocker 

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM|ARM|IAR

**Expected behavior**

32 bit( int) return value from Timer::read_ms().

**Actual behavior**

Unexpected return value from Timer::read_ms() and read_high_resolution_us(),
'read_high_resolution_us()'  is return 64 bit and application using in 32 bit.

**Steps to reproduce**

 Test case tests-mbed_drivers-timer. 

```
 printf("p_timer.read()=%f\r\n",p_timer.read());   ---> OK 
 printf("p_timer.read()=%d\r\n",p_timer.read_ms()); --> NOT OK
 printf("p_timer.read()=%d\r\n",p_timer.read_us());---> OK
 printf("p_timer.read()=%d\r\n",p_timer.read_high_resolution_us()); --> NOT OK
```

----------------------------------------------------------------

**Suggested enhancement**

`typedef uint64_t us_timestamp_t;`
What is the usage of placing 64bit for us_timestamp_t while  us_ticker_read(void) return always 32bit.  
`uint32_t us_ticker_read(void)`

This issue got resolved when adding type conversion(64 bit to 32 bit) before returning "read_high_resolution_us()" function.

```
us_timestamp_t Timer::read_high_resolution_us() {
    core_util_critical_section_enter();
    us_timestamp_t time = _time + slicetime();
    core_util_critical_section_exit();
    return (int)time;   ---->type cast to int
}
```
## Question
Is my understanding is correct ? else how can I resolve this issue.

0xc0170, 2018-01-19:
>   printf("p_timer.read()=%d\r\n",p_timer.read_high_resolution_us()); --> NOT OK

This code is wrong, `us_timestamp_t` is not int as %d suggests, use proper type. Use `PRIu64` instead, or llu that should work I believe for all our 3 toolchains

akrnvm, 2018-01-19:
OK,
So below checking are fine.
    TEST_ASSERT_FLOAT_WITHIN(DELTA_S(4), 1.060f, p_timer->read());
    TEST_ASSERT_INT32_WITHIN(DELTA_MS(4), 1060, p_timer->read_ms());
    TEST_ASSERT_INT32_WITHIN(DELTA_US(4), 1060000, p_timer->read_us());
    TEST_ASSERT_UINT64_WITHIN(DELTA_US(4), 1060000, p_timer->read_high_resolution_us());

But, my test case is failing in below assertion without modification and it passing with modification (type casting)
  TEST_ASSERT_UINT64_WITHIN(DELTA_US(4), 1060000, p_timer->read_high_resolution_us());

0xc0170, 2018-01-19:
delta macro returns integer, not certain why.

> But, my test case is failing in below assertion without modification and it passing with modification (type casting)

What does this mean? what target what toolchain? How to reproduce?

cc @mprse

closed 2018-01-22 by akrnvm

cmonr, 2018-01-22:
@akrnvm Is there any particular reason why you closed the issue? Were you able to resolve it?

akrnvm, 2018-01-30:
Yes, now it's  working . I have modified my timer code and able to get ms and read_high_resolution_us value properly.