ARMmbed/mbed-os #5891: MemoryAPI documentation question

pauluap, 2018-01-19:
### Description
- Type: Question
- Related issue: `#abc`
- Priority: Blocker | Major | Minor

## Question

I opened a Stackoverflow issue regarding my question - [StackOverflow](https://stackoverflow.com/questions/48335691/mbed-memorypool-is-it-usable-for-nontrivial-classes)

Basically, it is not completely clear to me whether if the MemoryPool API is usable with non-POD classes that have non-trivial constructors.  

Either way, could the documentation be enhanced as to specify what types is usable with the MemoryPool API?

It seems to me that if I want to use a memory pool for nontrivial classes I would need to create one myself or follow one of the other suggestions in the answers/comments to the StackOverflow question.

0xc0170, 2018-01-22:
> Either way, could the documentation be enhanced as to specify what types is usable with the MemoryPool API?

I could not locate it neither, we will review it.

> It seems to me that if I want to use a memory pool for nontrivial classes I would need to create one myself or follow one of the other suggestions in the answers/comments to the StackOverflow question.

The memory pool API is using memory pool API from RTOS that is in C. The functions you look for if you want to see the implementation are `svcRtxMemoryPoolNew` and `osRtxMemoryPoolAlloc` (if pool is alloc is enabled in rtos, it is in our case).

@SenRamakri Please review

SenRamakri, 2018-01-29:
The MemoryPool API is purely a pool based memory allocator built on top of CMSIS/RTX Memory Pool implementation(https://www.keil.com/pack/doc/CMSIS/RTOS2/html/group__CMSIS__RTOS__PoolMgmt.html). It does not invoke the object constructor for the objects placed in the pool blocks. The memory pool allocates blocks of memory from the pool and returns the pointer of that block to the user. Its the responsibility of application to initialize the objects placed in memory blocks as required. Hope this clarifies, we will update the documentation on MemoryPool with this info.

0xc0170, 2018-01-30:
https://github.com/ARMmbed/Handbook/pull/388 - should fix this issue

AnotherButler, 2018-01-30:
PR #388 has merged, which I believe solves this issue. Please feel free to reopen if you do not feel this has been adequately resolved.

closed 2018-01-30 by AnotherButler