ARMmbed/mbed-os #5893: LoRa L-Tek FF1705_L151CC target name is incorrect

gggekov, 2018-01-22:
**Description**
**Type:** Documentation Bug
**Priority:** Minor
**Target:**FF1705_L151CC ( https://os.mbed.com/platforms/L-TEK-FF1705/)

**Toolchain:** GCC_ARM

**mbed-cli version:** 1.1.1

**Behavior**
The documentation says the target name for the board should be **FF1705_L151CC**. However,this target name is unrecognised by mbed-cli and the correct target name is **XDOT_L151CC**. Please update the documentation to the correct target name.

gorazdko, 2018-01-24:
> However,this target name is unrecognised by mbed-cli

It is recognised. Use the latest version of mbed-os 

```
mbed test -t ARM -m UNKNOWN
WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
usage: test.py [-h] [-m MCU] [-t TOOLCHAIN] [--color] [--cflags CFLAGS]
               [--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
               [--profile PROFILE] [--app-config APP_CONFIG] [-D MACROS]
               [-j JOBS] [--source SOURCE_DIR] [--build BUILD_DIR] [-l]
               [-p PATHS] [-f FORMAT] [--continue-on-build-fail] [-n NAMES]
               [--test-config TEST_CONFIG] [--test-spec TEST_SPEC]
               [--build-report-junit BUILD_REPORT_JUNIT]
               [--build-data BUILD_DATA] [-v] [--stats-depth STATS_DEPTH]
test.py: error: argument -m/--mcu: UNKNOWN is not a supported MCU. Supported MCUs are:
ARCH_BLE,              ARCH_BLE_BOOT,         ARCH_BLE_OTA,
ARCH_GPRS,             ARCH_LINK,             ARCH_LINK_BOOT,
ARCH_LINK_OTA,         ARCH_MAX,              ARCH_PRO,
ARM_BEETLE_SOC,        ARM_CM3DS_MPS2,        ARM_IOTSS_BEID,
ARM_MPS2_M0,           ARM_MPS2_M0P,          ARM_MPS2_M1,
ARM_MPS2_M3,           ARM_MPS2_M4,           ARM_MPS2_M7,
B96B_F446VE,           BLUEPILL_F103C8,       DELTA_DFBM_NQ620,
DELTA_DFCM_NNN40,      DELTA_DFCM_NNN40_BOOT, DELTA_DFCM_NNN40_OTA,
DELTA_DFCM_NNN50,      DELTA_DFCM_NNN50_BOOT, DELTA_DFCM_NNN50_OTA,
DISCO_F051R8,          DISCO_F100RB,          DISCO_F303VC,
DISCO_F334C8,          DISCO_F401VC,          DISCO_F407VG,
DISCO_F413ZH,          DISCO_F429ZI,          DISCO_F469NI,
DISCO_F746NG,          DISCO_F769NI,          DISCO_L053C8,
DISCO_L072CZ_LRWAN1,   DISCO_L475VG_IOT01A,   DISCO_L476VG,
EFM32GG_STK3700,       EFM32HG_STK3400,       EFM32LG_STK3600,
EFM32PG12_STK3402,     EFM32PG_STK3401,       EFM32WG_STK3800,
EFM32ZG_STK3200,       ELEKTOR_COCORICO,      ELMO_F411RE,
EV_COG_AD3029LZ,       EV_COG_AD4050LZ,       FF1705_L151CC,
FF_LPC546XX,           HEXIWEAR,              HRM1017,
```

0xc0170, 2018-01-24:
I can also see `FF1705_L151CC` in targets.json file. What is missing the report is SHA or version of mbed OS that this was tested against. I've just tested latest master.

gggekov, 2018-01-24:
I am getting this problem when working on a project using `mbed-os-5.4`, commit id is `fc1836545dcc2fc86f03b01292b62bf2089f67c3`.

0xc0170, 2018-01-24:
This target was added https://github.com/ARMmbed/mbed-os/pull/4973 (5.6.1). Please update the mbed OS to get the target support

This should be resolved, I'll close

closed 2018-01-24 by 0xc0170