ARMmbed/mbed-os #5916: STM32L0 wrong UART interrupt index

marcemmers, 2018-01-24:
### Description
- Type: Bug
- Related issue:
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
STM32L073RZ

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-os sha:**
96d9a00d0a1d25095b330095fa81c40f7741777c

**Actual behavior**
Mismatch on index used for interrupt handling. This hangs systems that only has UART4, UART5 or LPUART1 initialized on reception of a byte.

For example LPUART1:
The index is set to 4 in the STM serial_api.c.
The lpuart1 interrupt is mapped to index 2 in the STM32L0 serial_device.c
Because the interrupt flag is not cleared in the interrupt stays pending causing the system to hang. Even if multiple UARTS were initialized the reception handling would fail.

**Possible solution**
Move the LPUART1 to the last index in serial_device.c so USART4 gets idx 2, USART5 gets idx 3 and LPUART1 gets idx 4. See STM32L4 serial_device.c

0xc0170, 2018-01-24:
@ARMmbed/team-st-mcd Please review

bcostm, 2018-01-24:
It looks like the same issue as fixed in PR #5907

marcemmers, 2018-01-24:
It is indeed in the same line of the issue fixed in PR https://github.com/ARMmbed/mbed-os/pull/5907

I have created a separate PR for this issue: https://github.com/ARMmbed/mbed-os/pull/5917

bcostm, 2018-02-01:
ST_TO_BE_CLOSED

closed 2018-02-01 by 0xc0170