ARMmbed/mbed-os #5919: NXP LPC4088 Type I IOCON HS bit

scamille, 2018-01-24:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug / Enhancement
- Related issue: `#5918`
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NXP|LPC4088

**Toolchain:**
GCC_ARM

**Toolchain version:**
gcc version 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437] (GNU Tools for ARM Embedded Processors 6-2017-q2-update)

**mbed-cli version:** 
1.2.2

**mbed-os sha:**
97e2d4a8c (HEAD -> master) Merge pull request #5025 from grygorek/master

**DAPLink version:**

**Expected behavior**
When using Pins of the Type I IOCON register (P0[27], P0[28], P5[2], P5[3]) as normal GPIO pins (not for specialized I2C functions), as well as for SSP2, they should behave as normal GPIO pins as described in chapter 7.3.8 of UM10562 (https://www.nxp.com/docs/en/user-guide/UM10562.pdf). This means the HS bit ([8:8]) should be set to 1 when configuring the pin, somewhere in pinmap_pinout, pin_function or pin_mode

**Actual behavior**
SSP2 with P5[2] and P5[3] is currently not usable without special code setting the HS bit, e.g.

```
  for( auto&& pin : {sclk, ssel})
  {
    auto reg = (__IO uint32_t*) (LPC_IOCON_BASE + 4 * pin);

    // Type I IOCON register pin HS bits: [8:8] -> b1'0000'0000 = (0x100)
    *reg = (*reg & ~0x100) | (1 << 8); // set HS bit to 1
  }
```

**Comments**
Since I cannot estimate the impact of any hacky changes to eg. pin_mode, I cannot offer a ready-made pull-request here. The thing is that the solution depends both on the function (!I2C) and the type of the IOCON Base of the pin, and requires some extended logic.

0xc0170, 2018-01-24:
@ARMmbed/team-nxp 

@toyowata

scamille, 2018-01-26:
While stumbling some more around the UM10562 document, I notice that at page 119 on P5[3] it states:

"SSP2_SSEL — Slave select for SSP2. When using this pin, the SSP2 bit rate is
limited to 1 MHz."

Since I did resolve my SSP2 problems with the above described fixes to HS bit of SCLK pin P5[2] and SSEL pin P5[3] at 5MHZ SPI Clock, I suspect that the whole thing is only relevant for SCLK > 1MHz.

So depending on whether it is important for the mbed project to support SPI Slave mode on SSP2 on this board, this Issue could also be dropped if it is not worth the effort.

0xc0170, 2019-08-14:
@scamille Have you resolved the issue?

scamille, 2019-09-28:
I no longer work on this project. But back then I was able to resolve it by manually setting the HS bit, thus supporting >1MHZ SPI Clock speed. I cannot 100% assess if this behaviour is something you want enabled by default or not.

This ticket was mainly intended to report the problem to the mbed team for you to discuss if you want to handle it. Sorry I could not offer any patch to directly fix it.

Feel free to close the issue if it does not have any priority on your side. As said, for me personally it is no longer relevant.

0xc0170, 2019-09-30:
Thanks for the update. 

> So depending on whether it is important for the mbed project to support SPI Slave mode on SSP2 on this board, this Issue could also be dropped if it is not worth the effort.

👍

closed 2019-09-30 by 0xc0170