ARMmbed/mbed-os #5921: nsapi_dns_query_multiple() is broken

kFYatek, 2018-01-24:
### Description
- Type: Bug
- Priority: Major

All the public overloads of `nsapi_dns_query_multiple()` are broken. They are [documented](https://github.com/ARMmbed/mbed-os/blob/f1cf77fa443ee48b29861a8581f5d3f685c7ec2e/features/netsocket/nsapi_dns.h#L114) as returning the number of addresses found on success. However, the return value of `dns_scan_response()` is not passed down, [only its sign is checked](https://github.com/ARMmbed/mbed-os/blob/f1cf77fa443ee48b29861a8581f5d3f685c7ec2e/features/netsocket/nsapi_dns.cpp#L246) and the function always returns `NSAPI_ERROR_OK`, i.e. zero.

This is especially disastrous for all the overloads that take `SocketAddress *` as argument; [this overload](https://github.com/ARMmbed/mbed-os/blob/f1cf77fa443ee48b29861a8581f5d3f685c7ec2e/features/netsocket/nsapi_dns.cpp#L275) expects the static variant to work as documented, which causes none of the addresses to be copied.

betzw, 2018-01-24:
Nice catch!

0xc0170, 2018-01-24:
@kjbracey-arm @mikaleppanen Please review

mikaleppanen, 2018-01-26:
@betzw Yes, that is true. Do you want so submit a patch for this?

kFYatek, 2018-01-26:
@mikaleppanen I can write a patch just fine, but I haven't signed the CLA (yet?) so I'm not sure if I'm allowed to (for now). I don't know how long that whole legal process is... required changes seem pretty trivial so it might be faster if someone else does it.

betzw, 2018-01-26:
@kFYatek, @mikaleppanen,
Pls. see PR #5946

kjbracey-arm, 2018-01-26:
@kFYatek Can you confirm that #5945 fixes all your (dns-related) problems?

kFYatek, 2018-01-26:
Yes, looks fine, as I commented there.

PR #5945 (2018-01-29): Correct return value of nsapi_dns_query_multiple
Documentation states that `nsapi_dns_query_multiple` returns the number of
addresses found on success - it was returning 0.

Overloads using `SocketAddress` are relying on the return value, meaning
those calls didn't work at all.

Fixes #5921.
commit 15a3922f (2018-01-29): Correct return value of nsapi_dns_query_multiple

Documentation states that nsapi_dns_query_multiple returns the number of
addresses found on success - it was returning 0.

Overloads using SocketAddress are relying on the return value, meaning
those calls didn't work at all.

Fixes #5921.
commit 6e2584e3 (2018-02-09): Correct return value of nsapi_dns_query_multiple

Documentation states that nsapi_dns_query_multiple returns the number of
addresses found on success - it was returning 0.

Overloads using SocketAddress are relying on the return value, meaning
those calls didn't work at all.

Fixes #5921.
closed 2018-01-29 by cmonr