ARMmbed/mbed-os #5924: nRF52840 Custom Board hangs if printf() is used within a loop

vergil-SI, 2018-01-24:
### Description
- Type: Bug 
---------------------------------------------------------------
## Bug

**Target**
nRF52840: Custom Board

**Toolchain:**
GCC_ARM

**Toolchain version:**
2017 q2
**mbed-cli version:** 
1.3.0

**Expected behavior**
Device can print the same message over and over within a loop.
**Actual behavior**
Device hangs.

## Question

Hi, I'm working with a custom board that uses the nRF52840_xxAA. 

Before jumping in, I first set up a development environment with the nRF52840_PDK (it has the same silicon revision), using mbed-cli, and eclipse. I exported to gnuarmeclipse so that I could build inside of eclipse using "mbed compile" for easy debugging. I have been able to get the PDK up and running, with little to no issues.

Using the PDK as a reference, I created a custom target in mbed-os by adding the board into a custom_targets.json, and giving it its own PinNames, header file. I'm able to get the LED on the board blinking, and the board seems to behave correctly, up until I attempt to use the UART. Specifically, I can use the UART, until I attempt to printf() inside a loop. When I do so, the board functions for approximately around three seconds, before it hangs. (this is independent of how may printfs I do in the loop, if I lower the wait time between them, I am able to do more printfs until the board hangs, and vice versa.) I'm confused as to what could be causing this issue since I'm using the same code on the same chip. Could there be something I specifically need to define involving UART for the custom target?

I'm using the same TX and RX pins (6 and 8) as the PDK, and am just looking for some insight to see if anyone else has experienced a similar issue, or knows of something that could cause such this kind of problem.

jrobeson, 2018-01-24:
did you need to turn on/off hardware flow control?

vergil-SI, 2018-01-25:
I had thought that could have something to do with it, but have yet to see any results by changing it. To enable it and disable it is as simple as modifying the mbed_config.h correct? (The document says DO NOT EDIT though, is there a more proper place to turn this on or off? I couldn't find a parameter that sounded right for it in the targets.json, or elsewhere)

Also, a further update, if I connect the RX on the chip to the TX line of my FTDI cable, the board doesn't function AT ALL. I had previously only been connecting the TX of the board . The flow control issue sounds even more viable after this behavior.

vergil-SI, 2018-01-25:
Upon further review, of the targets.json, I found the macro under MCU_NRF52840, and disabled it, and the board is working great! I had expected something like that to be set under the PDK itself rather than the chip as a whole.

closed 2018-01-26 by vergil-SI