ARMmbed/mbed-os #5930: Serial using PA_11 & PA_12 stops working when using the default serial pins (PA_2 or PA_3) as analog or digital in/outs.

damoww, 2018-01-25:
## Bug

Serial using PA_11 & PA_12 stops working when using the default serial pins (PA_2 or PA_3) as analog or digital in/outs.
---------------------------------------------------------------
**Target**
Nucleo F401RE

**Toolchain:**
GCC_ARM  ||  online compiler

**mbed-os version:** 
Bug occurs _after_ 5.4.3

**SHA for commit which breaks my code**
d740bde646e098f503a6d17e24674c68196ee2da

**Expected behavior**
Serial works using pins PA_11 & PA_12 even when using the default serial pins (PA_2 or PA_3) as analog or digital in/outs

**Actual behavior**
Serial stops working when using pins PA_11 & PA_12 _while_ the default serial pins (PA_2 or PA_3) are used as analog or digital in/outs.

**Details**

I tried to upgrade our mbed OS version from 5.4.3 to the latest. After doing so all serial communication to the pc stopped working. 

I am not using the default pins (PA_2 & PA_3) for Serial, instead I am using them for AnalogIn to suit our PCB layout. For Serial I am using `PA_11` & `PA_12`. I have done the correct modifications to my Nucleo board to suit this change and both AnalogIn and Serial have been working as expected - until I upgraded from 5.4.3.

`#include "mbed.h"

// DigitalOut led1(PA_3);
// AnalogIn led2(PA_2);

Serial serial(PA_11, PA_12, 921600);

int main() {
    while (true) {
        serial.printf("HELLLO\r\n");
        wait(0.5);
    }
}`

After upgrading my mbed OS version, the above code will still work. When I remove the comment next to either `led1` or `led2` the Serial will stop working.

I think I have located the commit which creates this bug in 5.4.4 onwards:
**[[STM32F4] Remove SERIAL_TX and SERIAL_RX from available pins](https://github.com/ARMmbed/mbed-os/commit/d740bde646e098f503a6d17e24674c68196ee2da#diff-ae49ea6b4ffe16a75a3a10d2b0fcb96b)**

The file:
`mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F401xE/TARGET_NUCLEO_F401RE/PeripheralPins.c`

0xc0170, 2018-01-25:
@ARMmbed/team-st-mcd Please review

jeromecoutant, 2018-01-25:
Hi
I agree that PA_2 and PA_3 are hardcoded as non-available...

For a quick solution, uncomment the lines in the PeripheralPins.c file

jeromecoutant, 2018-01-25:
@damoww could you check https://github.com/ARMmbed/mbed-os/pull/5934 ?

jeromecoutant, 2018-01-26:
#5934 is now merged in 5.7.4

cmonr, 2018-01-26:
Closing due to PR being merged.

closed 2018-01-26 by cmonr

damoww, 2018-01-27:
FYI, if I uncomment the lines in the PeripheralPins.c file everything works as expected.
Cheers for the quick responses and getting a solution into master 😄