ARMmbed/mbed-os #5931: STM serial_init change broke STM target interrupts

jarvte, 2018-01-25:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Blocker
---------------------------------------------------------------
## Bug

**Target**
STM

**Toolchain:**
GCC_ARM|ARM|IAR

**mbed-os sha which caused the issue:**
https://github.com/ARMmbed/mbed-os/commit/62a7ecddd356293a3f4f0557ce41e4e66d08353d

**Expected behavior**
Interrupt indexes should match.

**Actual behavior**
In file targets/TARGET_STM/serial_api.c there is a function void serial_init(serial_t *obj, PinName tx, PinName rx) which was moved from all the target/TARGET_STM/xxx/serial_device.c’s to reduce code.
It’s behaviour was also changed so that obj_s->index is not anymore hardcoded value but using IndexNumber and incrementing that.
That raises a problem as target/TARGET_STM/TARGET_STM32F4/serial_device.c still uses the hard-coded values and so they don’t match.

jarvte, 2018-01-25:
@jeromecoutant @theotherjimmy

0xc0170, 2018-01-25:
@ARMmbed/team-st-mcd  Please review

jeromecoutant, 2018-01-25:
Hi
I agree that in TARGET_STM32F4/serial_device.c file, we should use an incremental index like in serial_api.c

marcemmers, 2018-01-25:
This is the same issue as we have solved in PR https://github.com/ARMmbed/mbed-os/pull/5907 and PR https://github.com/ARMmbed/mbed-os/pull/5917

See PR 5917 for an incremental implementation for the STM32L0

bcostm, 2018-01-25:
To avoid code duplication, I think the new function `uart_pointer_get_index` created by @marcemmers for the L0 must be placed in serial_api.c file. So, we can re-use it for all STM32 devices.

0xc0170, 2018-01-25:
> To avoid code duplication, I think the new function uart_pointer_get_index created by @marcemmers for the L0 must be placed in serial_api.c file. So, we can re-use it for all STM32 devices.
> I'll do it.

Thanks @bcostm 💯

marcemmers, 2018-01-25:
@bcostm might as well improve it a little bit by at least making it an inline function.

screamerbg, 2018-01-31:
@bcostm Any update on this. Please note that this impact other partner's targets as well, e.g. u-blox, Multitech, Seeed. At this point this becomes a show-stopper for connectivity testing as well, like expansion modules, shields, etc.

marcemmers, 2018-01-31:
@screamerbg see #5962. It is ready for merge as of an hour ago.

bcostm, 2018-02-01:
ST_TO_BE_CLOSED

closed 2018-02-01 by 0xc0170