ARMmbed/mbed-os #5935: How to build LittleFileSystem on Keil

olavipartanen, 2018-01-25:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Question

---------------------------------------------------------------
## Bug

**Target**
STM32L4

**Toolchain:**
Keil

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**mbed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

0xc0170, 2018-01-25:
What's your mbed OS version (SHA also might do) ? What are you exactly building ?

What toolchain version do you use?

From the code, seems this function is causing a trouble
```
static inline uint32_t lfs_ctz(uint32_t a) {
#if defined(__GNUC__) || defined(__CC_ARM)
    return __builtin_ctz(a);
#elif defined(__ICCARM__) && defined(__CLZ)
    return __CLZ(__RBIT(a));
#else
    uint32_t r = 32;
    a &= -a;
    if (a) r -= 1;
    if (a & 0x0000ffff) r -= 16;
    if (a & 0x00ff00ff) r -= 8;
    if (a & 0x0f0f0f0f) r -= 4;
    if (a & 0x33333333) r -= 2;
    if (a & 0x55555555) r -= 1;
    return r;
#endif
}
```

@geky Please review

geky, 2018-01-25:
I'm having a hard time reproducing this error. I exported mbed-os using `mbed export -i uvision -m NUCLEO_L476RG`, but the project was able to compile without an issue.

The Keil documentation online indicates that `__builtin_ctz` is provided by the compiler, so it's strange that you aren't able to compile.
http://www.keil.com/support/man/docs/armcc/armcc_chr1359125006834.htm

Some questions:
- How did you compile the project?
- What board are you compiling for?
- Did you use `mbed export` to generate the Keil project?
- What version of uvision are you using? (you can find this in: -> help -> about uvision)

As a work around you can use the software implementation of `lfs_ctz` by commenting out the other options, though I'd be very interested to know what's causing this compilation issue.

olavipartanen, 2018-01-26:
> Lähetetty: Thursday, January 25, 2018 10:56 PM
> Vastaanottaja: ARMmbed/mbed-os <mbed-os@noreply.github.com>
> Kopio: olavipartanen <olavi.partanen@pollentech.io>; Author <author@noreply.github.com>

Hi,

 

Thank you for the quick response.

 

I have STM32Cube project for STM32L4R9i-Eval board and I use Keil uVision V5.24.2.0 with “Default compiler version 5” selected from “ARM compiler”.

Yes, I also find out that __builtin_ctz function is from compiler, so it seems that Keil’s compiler isn’t applicable. 

 

“As a work around you can use the software implementation of lfs_ctz by commenting out the other options” – I found that lfs_ctz is called from lines 1067 and 1142 in lfs.c but I don’t really know how to replace those calls or __builtin_ctz  call in lfs_util.h.

 

 

-Olavi Partanen

http://www.keil.com/support/man/docs/armcc/armcc_chr1359125006834.htm

olavipartanen, 2018-01-26:
Hi,

 

I had lfs from https://github.com/geky/littlefs and just realized that ARMmbed version of lfs is little bit different. 

I found “other options” there, so with this “work around” by commenting out __builtin_ctz I was able to compile the lfs.

 

Still thinking that is there a way to use other compiler in Keil or build lfs from cmd the way what can be include to Keil compilers build.

 

-Olavi

 

Lähettäjä: Olavi Partanen [mailto:olavi.partanen@pollentech.io] 
Lähetetty: Friday, January 26, 2018 8:29 AM
Vastaanottaja: 'ARMmbed/mbed-os' <reply@reply.github.com>
Aihe: VS: [ARMmbed/mbed-os] How to build LittleFileSystem on Keil (#5935)

 

Hi,

 

Thank you for the quick response.

 

I have STM32Cube project for STM32L4R9i-Eval board and I use Keil uVision V5.24.2.0 with “Default compiler version 5” selected from “ARM compiler”.

Yes, I also find out that __builtin_ctz function is from compiler, so it seems that Keil’s compiler isn’t applicable. 

 

“As a work around you can use the software implementation of lfs_ctz by commenting out the other options” – I found that lfs_ctz is called from lines 1067 and 1142 in lfs.c but I don’t really know how to replace those calls or __builtin_ctz  call in lfs_util.h.

 

 

-Olavi Partanen

Vastaanottaja: ARMmbed/mbed-os <mbed-os@noreply.github.com <mailto:mbed-os@noreply.github.com> >
Kopio: olavipartanen <olavi.partanen@pollentech.io <mailto:olavi.partanen@pollentech.io> >; Author <author@noreply.github.com <mailto:author@noreply.github.com> >

 

http://www.keil.com/support/man/docs/armcc/armcc_chr1359125006834.htm

olavipartanen, 2018-01-26:
Hi,

 

I don’t use mbed, I have Keil uVision project with default compiler.

 

It seems that __builtin_ctz function is from compiler and it’s not available in Keil’s compiler. So as a work around I commented out __builtin_ctz function calls. 

(thanks for those to Christopher Haster who also answered to my issue)

 

I’ll try if this works.

 

Thank you.

 

-Olavi

Lähetetty: Thursday, January 25, 2018 5:15 PM

geky, 2018-01-28:
Hi @olavipartanen,

I managed to get my hands on Keil to try a few things and think I've figured out the issue. It looks like mbed-os is using the flag `--gnu` for Keil. For some reason, this changes when `__builtin_ctz` is emitted, but not `__builtin_clz` or `__builtin_popcount`.

So the current version should compile if you provide the `--gnu` flag.

I've pushed up a patch to not use `__builtin_ctz` and provide the software implementations when necessary (https://github.com/geky/littlefs/commit/53f9834b1deb342ed5810f333b23d06dfd6785f7). I'll merge it into master as soon as it passes testing.

Thanks for reporting this!

cmonr, 2018-02-01:
@olavipartanen If @geky's answer doesn't quite solve the issue, feel free to reopen.

closed 2018-02-01 by cmonr

geky, 2018-02-22:
Just an fyi, this has since been fixed in littlefs mainline:
https://github.com/geky/littlefs/commit/4f08424b51c6e7fda140b91fd551a35531ca0fc3